Ultimaker/Cura #2428: Cura 2.7 crashes at startup on fedora 26

vrubiolo, 2017-09-14:
Application Version: Cura 2.7 (AppImage)
Platform: Linux Fedora 26
Qt: not sure (I think 
PyQt: << The version of PyQt used >>
Display Driver: Wayland

Steps to Reproduce:
./Cura-2.7.0.AppImage or double-click on the Cura 2.7 icon

Actual Results:
Cura starts up, the splash screen appears and gets updated to the 'Loading machines' part. Then it stays there, I hear my CPU fan speed going up for several seconds, a black window appears for 1/10th of a second and the program crashes.

Expected results:
Cura should start normally.

Additional Information:
Terminal output:
$ ./Cura-2.7.0.AppImage 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_znbNGW/usr/bin/resources/qml/JobSpecs.qml:160:13: QML TooltipArea: Binding loop detected for property "height"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
./bin//cura.sh: line 11:  3567 Segmentation fault      (core dumped) cura $@

Output of journalctl after the crash:
Sep 13 22:46:21 flatman systemd-coredump[3603]: Process 3567 (cura) of user 1000 dumped core.
                                                
Stack trace of thread 3567:
#0  0x00007fa9fb3d925b raise (libpthread.so.0)
#1  0x00007fa9fb3d93b0 n/a (libpthread.so.0)
                                                
Stack trace of thread 3573:
#0  0x00007fa9fa7d7d03 __select (libc.so.6)
#1  0x00007fa9f2be8832 n/a (/tmp/.mount_znbNGW/usr/bin/lib/python3.5/select.cpython-35-x86_64-linux-gnu.so)
#2  0x000000000002f48f n/a (n/a)
Sep 13 22:46:21 flatman audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=systemd-coredump@3-3602-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostna
Sep 13 22:46:23 flatman abrtd[876]: Size of '/var/spool/abrt' >= 5000 MB (MaxCrashReportsSize), deleting old directory 'ccpp-2017-09-13-22:40:05.70368-897'
Sep 13 22:46:24 flatman abrt-notification[3663]: Process 3567 (cura) crashed in __restore_rt()

Cura GUI log is attached.

vrubiolo, 2017-09-14:
[cura.log.txt](https://github.com/Ultimaker/Cura/files/1301851/cura.log.txt)

nallath, 2017-09-14:
Hmm, I'm still on fedora 25 and there it works.

vrubiolo, 2017-09-14:
Good to know, I'll see if maybe a docker container would do to circumvent the issue.

ericgibert, 2017-09-17:
Same problem on Fedora 25:

./Cura-2.7.0.AppImage 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_5Jlygi/usr/bin/resources/qml/JobSpecs.qml:160:13: QML TooltipArea: Binding loop detected for property "height"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
./bin//cura.sh: line 11: 24848 Segmentation fault      (core dumped) cura $@

ericgibert, 2017-09-17:
Note that the packaged Cura 15.04.4 works fine of Fedora 25 (installed with yum).

Samon33, 2017-09-18:
I'm experiencing this same issue (on Fedora 26) with several different versions of the 2.x branch. As ericgilbert noted, the packaged 15.x branch works fine.

I recently rebuilt my machine on F26, but on F25, I was able to run several different AppImage packaged versions (I have 2.5, 2.6 and 2.7 downloaded), as well as the builds from Churchyard's COPR repository (https://copr.fedorainfracloud.org/coprs/churchyard/cura-modern/) - all of these exhibit the issue described here on F26.

vrubiolo, 2017-09-18:
Thanks for the feedback. I confirm I have 15.04.6 working fine on this machine. 
Note that I said I am running using Wayland on F26 but this is actually with the Xorg compability layer enabled (either X11 on Wayland or X11 directly, not sure) as I cannot run 15.04.6 with pure Wayland (it also crashes, this was happening for me on Fedora 25 too). With X11 compability enabled 15.04.6 works fine but 2.7 crashes.

nallath, 2017-09-18:
That 15.04 works doesn't say anything. It has completely different code and libraries that it depends on. For example; 15.04 uses wxWidgets, whereas cura 2.x uses Qt as a graphics framework. 

It's really weird that this doesn't work. I develop on Fedora myself and I don't have this issue. Could be because I have certain libraries a bit different, but even the app image doesn't cause issues.

vrubiolo, 2017-09-18:
@nallath : thanks for the feedback, I was mentioning Wayland/X11 to give precisions on my graphical stack/setup.  I did not know the graphical library was different between 15.x and 2.x. 
Glad to see you are on Fedora too, this will make solving this easier. Is there a way we can gather more information for you to investigate (I can collect gdb/pdb backtraces, etc if needed)?

nallath, 2017-09-19:
Yeah, I'm afraid we are at the point of needing backtraces. If you could provide them, that would be really helpful.

vrubiolo, 2017-09-19:
Ok great. Are the debugging symbols included in the AppImage for the debugger to use? Also, are there available instructions on debugging Cura as an AppImage (as it is currently packaged)?

vrubiolo, 2017-09-20:
[cura-2.7-fc26-crash-bt.txt](https://github.com/Ultimaker/Cura/files/1316675/cura-2.7-fc26-crash-bt.txt)

nallath, 2017-09-20:
I always run from source, so for me it's using `PYTHONPATH=.:/usr/local/lib/python3.5/site-packages:../Uranium:/usr/local/lib64 gdb -ex r --args python3 cura_app.py`

As far as I can see now, it's a segfault with Qt somewhere. I'm probably not on the exact same version of Qt / PyQt. On ot the other hand; the app image should contain everything and that does work for me. 

Does the 2.7.0-beta or 2.6.2 work?

vrubiolo, 2017-09-21:
Same crash, seemingly at the same place for both 2.6.2 and 2.7.0-BETA.
Here are the traces and the cura logs: 
* 2.6.2 - [backtrace_Cura-2.6.2.txt](https://github.com/Ultimaker/Cura/files/1319758/backtrace_Cura-2.6.2.txt) & [cura-2.6.2.log.txt](https://github.com/Ultimaker/Cura/files/1319757/cura-2.6.2.log.txt)
* 2.7.0-BETA - [backtrace_Cura-2.7.0-BETA.txt](https://github.com/Ultimaker/Cura/files/1319759/backtrace_Cura-2.7.0-BETA.txt) & [cura-2.7.0-BETA.log.txt](https://github.com/Ultimaker/Cura/files/1319760/cura-2.7.0-BETA.log.txt)

nallath, 2017-09-21:
The one thing that could be an issue is your graphics drivers (as what else can it still be; I'm on fedora 25 with 2.7 source & app image and both work fine).

awhiemstra, 2017-09-21:
> #1  0x00007fffde05a600 in QSGRenderContext::initialize(QOpenGLContext*) () from /home/vincent/Documents/Dev/3D_Printing/Cura-2.7/Cura-2.6.2/extracted/usr/bin/libQt5Quick.so.5

This suggests something goes wrong when creating the OpenGL context. This might be due to issues with your driver or due to issues when using OpenGL from a Wayland X11 context. To be fair, we so far have not tested Cura on Wayland, so it having issues is not entirely unexpected.

vrubiolo, 2017-09-22:
@nallath : this is the onboard GPU of my Intel CPU (Intel® HD Graphics 5500 (Broadwell GT2), from the GNOME properties).
@awhiemstra : is there any more information that I can provide to diagnose this issue? As I explained before, I am choosing the 'GNOME on Xorg' option at my login manager at startup. I am interested in getting this solved/getting a workaround so that I can run (I have never run Cura 2.x so far because of that, it used to crash too when I had tried it around the 2.2 times IIRC).

wsmirnow, 2017-10-31:
Hi,
I've same issue on my laptop (IGPU) and workstation (NVIDIA) running Fedora 26. Cura crash with a segfault on start. 

The solution is to install libglvnd-devel package or create a symbolic link 
```sh
cd/usr/lib64
sudo ln -s libGL.so.1 libGL.so
```

Ghostkeeper, 2017-11-07:
@awhiemstra I've heard this about libglvnd-devel before. Is that something we should add to our build environment and include in the Linux build?

wsmirnow, 2017-11-09:
I wouldn't add the lib to the build because it is more or less distribution/hardware specific. I would prefere to create a spec script for rpm builds. This is actually done by churchyard/cura-modern COPR repo owner. Let me talk to him, if he would contribute it.

awhiemstra, 2017-11-09:
libglvnd is the OpenGL vendor-neutral dispatch library and is basically an attempt to fix what OpenGL library is going to be used when loading OpenGL. It is mostly relevant when having NVidia's driver installed, since most of the other drivers rely on Mesa as OpenGL implementation. I would definitely not recommend including it in the build as it is too low level for what we need - which is a working OpenGL implementation. In the end it sounds like this is a platform/driver issue. Having a workaround would be nice, but I do not think we should change much on our side.

wsmirnow, 2017-11-09:
I agree with that

closed 2018-03-02 by ChrisTerBeke