Ultimaker/Cura #2429: Messy G-code of Skin due to M204 and M205 commands

konvoj, 2017-09-14:
The G-code generated as TYPE:SKIN is messy as hell and this is unnecessary. 

Example:
;TYPE:SKIN
G1 F1200 X132.351 Y93.743 E9.68272
M204 S625
M205 X6 Y6
G0 F3750 X132.571 Y94.458
M204 S500
M205 X5 Y5
G1 F1200 X128.763 Y90.65 E9.76249
M204 S625
M205 X6 Y6
G0 F3750 X127.526 Y89.908
M204 S500
M205 X5 Y5
G1 F1200 X132.656 Y95.038 E9.86996
M204 S625
M205 X6 Y6
G0 F3750 X132.661 Y95.537
M204 S500
M205 X5 Y5.................

M204 and M205 commands are setting accelerations and jerks after each movement, but the change is so subtle - S500 to S625 and X5 Y5 to X6 Y6 that I can't find any reason why this should be the way.

In the end, the G-code is very long, badly readable and has bigger filesize than would be necessary otherwise.

This is just an example that i came across, it can be, that some other features than SKIN do show the same behaviour. The only one I can exclude is the WALL-OUTER which is generated in a usual way.

Ghostkeeper, 2017-09-20:
This is because the jerk and acceleration of the travel moves is different from the jerk and acceleration of the skin lines. It's switching back and forth.

If you set the two jerks/accelerations to be the same, it shouldn't print those lines any more.

I don't think this is much of a problem. It makes slicing slower, but then again trying to detect whether a move is short enough that the acceleration/jerk doesn't really matter would also make slicing slower.

closed 2017-10-19 by Ghostkeeper

venuasg, 2019-12-25:
Hi,

;TYPE:SKIN
G1 F1200 X132.351 Y93.743 E9.68272

Where is this line picking from cura settings, F1200 

Thanks

Ghostkeeper, 2019-12-30:
The `F` parameter is in mm per minute. Divide it by 60 to get the mm per second that's being shown in the Cura interface. In your case, that's 20mm/s. Probably the Top/Bottom Speed setting or the initial layer speed setting combined with the Top/Bottom Speed setting.