Ultimaker/Cura #2430: Suggestion to the solidworks plugin

RichardB95, 2017-09-14:
I have made some modifications to the solidworks macro for exporting STL files to Cura. Currently every file that is sent to Cura has the same name (model_for_cura). This isn't very helpful when you want to print of an SD card. My fix fetches the part name and exports the file to cura using this part name.

This is the modified macro:

' ******************************************************************************
' C:\Users\s.edwards\AppData\Local\Temp\swx6644\Macro1.swb - macro recorded on 01/30/17 by s.edwards
' ******************************************************************************
Dim swApp As ISldWorks
Dim Part As Object
Dim tempFileName As String
Dim tempDir As String
Dim All As Boolean
Dim boolStatus As Boolean
Dim stlBinaryFormat As Boolean
Dim stlQuality As Integer
Dim swModel      As SldWorks.ModelDoc2

Dim myWS As Object

Dim curaVersion As String
Dim curaDir As String

Private Function EnumerateRegistryKeys(hive, key) As Collection
  Dim curaKeyNames As New Collection
  Dim idx As Integer
  idx = 1

  'get all Cura key names
  Set reg = GetObject("winmgmts://./root/default:StdRegProv")
  reg.EnumKey hive, key, arrSubKeys
  If Not IsNull(arrSubKeys) Then
    For Each subkey In arrSubKeys
      If InStr(1, subkey, "Cura ") = 1 Then
        curaKeyNames.Add subkey
        idx = idx + 1
      End If
    Next
  End If

  'sort the cura key names with the latest version first
  Dim i As Long, j As Long
  Dim vTemp As Variant

  'Two loops to bubble sort
  For i = 1 To curaKeyNames.count - 1
    For j = i + 1 To curaKeyNames.count
      If curaKeyNames(i) < curaKeyNames(j) Then
        'store the lesser item
        vTemp = curaKeyNames(j)
        'remove the lesser item
        curaKeyNames.Remove j
        're-add the lesser item before the
        'greater Item
        curaKeyNames.Add vTemp, vTemp, i
      End If
    Next j
  Next i

  Set EnumerateRegistryKeys = curaKeyNames
End Function

Sub main()
  
  Set swApp = Application.SldWorks
  Set swModel = swApp.ActiveDoc
  Set Part = swApp.ActiveDoc
  
  
  FilePath = swModel.GetPathName
  
  FileName = Replace(UCase(Dir(FilePath)), ".SLDPRT", "")

  tempDir = CreateObject("WScript.Shell").ExpandEnvironmentStrings("%Temp%")
  tempFileName = tempDir & "\" & FileName & ".STL"
  ' need to change this
 
  

  ' Back up the current STL options so that we can restore them later.
  stlBinaryFormat = swApp.GetUserPreferenceToggle(swUserPreferenceToggle_e.swSTLBinaryFormat)
  stlQuality = swApp.GetUserPreferenceIntegerValue(swUserPreferenceIntegerValue_e.swSTLQuality)

  swApp.SetUserPreferenceToggle swUserPreferenceToggle_e.swSTLBinaryFormat, True  ' Use binary STL
  swApp.SetUserPreferenceIntegerValue swUserPreferenceIntegerValue_e.swSTLQuality, swSTLQuality_Fine  ' Up the quality

  ' Figure out where Cura is installed.
  Const HKEY_LOCAL_MACHINE = &H80000002
  strKeyPath = "SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall"

  ' Get all possible Cura locations
  Dim curaDirNames As Collection
  Set curaDirNames = EnumerateRegistryKeys(HKEY_LOCAL_MACHINE, strKeyPath)
  If curaDirNames.count = 0 Then
    MsgBox "Cannot find Cura on this computer.", vbInformation + vbOKOnly, "Cannot find Cura"
  End If
  Dim curaName As String
  curaName = curaDirNames.Item(1)  'use the first one which is the latest version

  ' Get Cura location
  Set myWS = CreateObject("WScript.Shell")
  curaDir = myWS.RegRead("HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall\" & curaName & "\DisplayIcon")
  ' remove the icon file
  Dim lastSlashIdx As Long
  lastSlashIdx = InStrRev(curaDir, "\")
  curaDir = Left(curaDir, lastSlashIdx)

  Part.ClearSelection2 (All)  ' Clear the selection. We want everything in the current document.
  Set Part = swApp.ActiveDoc

  Part.SaveAs tempFileName

  ' Restore the options
  swApp.SetUserPreferenceToggle swUserPreferenceToggle_e.swSTLBinaryFormat, stlBinaryFormat
  swApp.SetUserPreferenceIntegerValue swUserPreferenceIntegerValue_e.swSTLQuality, stlQuality

  Shell """" & curaDir & "Cura.exe"" --single-instance " & """" & tempFileName & """"

End Sub

nallath, 2017-09-14:
Could you please make a pull request for this? That makes it a lot easier for us to handle it (including giving proper attribution to you if it gets merged :))

thopiekar, 2017-09-15:
@RichardB95 I don't understand why you want to use the model name as the name for the temporary file? It only needs to be something unique, so a filename which is not being used at the moment.
To be honest, I would even remove all the exporting (into STL or whatever) from this macro. The plugin inside Cura already has the knowledge and the logic how to do that. Adding features in the macro or the plugin means the doubled amount of work to implement it on both.

@nallath Could we redirect all the people to https://github.com/Ultimaker/CuraSolidWorksPlugin/issues when it comes to SolidWorks plugin/macro issues? After adding more and more plugins it might get noisy here at Ultimaker/Cura :wink:

RichardB95, 2017-09-15:
@thopiekar  I print most of my files off a SD card. When I use the current plugin all the g-code is saved with the name model_for_cura (unless I save the g-code with a new name). This means that only the last model I sliced will be on the SD cards as all the ones would have been overwritten. If I wanted to print one of the old files I would have to go through the whole slicing process again.
What would be better is that when the model is brought into Cura with the file name of the solid works part. This means cura would save the g-code with that name without me having to go through the schlep of saving it with a new name. Then when I want to print an old part it is easy to find it on the SD card.
I have only looked at the current macro and it looks like creates a temp file anway:
  tempDir = CreateObject("WScript.Shell").ExpandEnvironmentStrings("%Temp%")
  tempFileName = tempDir & "\model_for_cura.STL"

What I added in was the following:
  FilePath = swModel.GetPathName
  FileName = Replace(UCase(Dir(FilePath)), ".SLDPRT", "")
  tempDir = CreateObject("WScript.Shell").ExpandEnvironmentStrings("%Temp%")
  tempFileName = tempDir & "\" & FileName

I created a fork and pull request on the Ultimaker/CuraSolidWorksPlugin repository

thopiekar, 2017-09-15:
@RichardB95 : Nice! I will take a closer look on this later. But however, as I mentioned before, we can open the sldprt or sldasm directly, since the plugin is already installed inside Cura.
What you did is almost the same like I did in the plugin itself.
For you to understand: The macro is a contribution by Ultimaker and has been developed in parallel because I kept believing in my way of opening files. Nevertheless, at the end we decided to bring everything together in one bundle.
You seem to be more familiar with macros, right? The reason I never review the macro is, that I wanted to open it, but was too busy and just waved the macro through into the plugin. Can you give some explanation?
Additionally: Are you familiar with plugins? Because macros seem to be only scripts.

thopiekar, 2017-11-09:
Hey @RichardB95,

Ultimaker shrinked now the macro to the minimum. Now the plugin is doing the conversion and is the only one, who is watching your settings.
Could you give it a try? At the moment a more stable version (0.3.7) is not uploaded to the plugin server. You can download and install the plugin manually from here: https://github.com/thopiekar/CuraSolidWorksPlugin/releases

closed 2017-11-09 by thopiekar