Ultimaker/Cura #2431: Second instance of Cura passively overwrites changed settings

MathijsG, 2017-09-14:
Maybe this is by design, but when I download STL files I often have multiple Cura windows opened, because everytime you double click a file it opens another instance of Cura.

And when I make a change in a certain instance of Cura and I close this instance of Cura earlier then another instance of Cura the setting gets set back. It only applies the setting change when I close this instance as the last. Is there some proces in the code that actively writes the settings of cura when you close the program, even when anything hasn't changed at all? It's quite confusing because you only discover this when you check at the certain setting and seeing it reverted back.

My proposal is to only rewrite settings on program quit when there has been a change, and maybe work with a time setting of that change so other instances of Cura won't revert older changes back.

I hope my issue is somewhat clear, if not I can provide a video.

nallath, 2017-09-14:
The problem with this is that if there is an unexpected close (eg; Cura crashes, It's forced closed by task manger, you get the idea), your solution means that the data is lost. 

Cura should only save profiles if it marked them as "dirty" (eg; something changed). Setting a different quality will mean that the printer gets saved again, but not that the qualities that are there get saved.

Ghostkeeper, 2017-09-20:
It always saves at closing down Cura though, even if nothing changed. In fact I think that the process of starting Cura causes the configuration to be marked as "dirty", because of all the profile changed signals going on.

I think it's expected that the last instance that was closed should be leading, even if no setting was changed in that instance since starting it. That could be just my programmer's mind speaking though, because I know the mechanism.

MathijsG, 2017-09-20:
@nallath but now the changed setting is lost everytime the cura instance where you did the change isn't the last Cura instance to close. I rather would have some loss at a crash, where it's quite common to lose some changes, then always having stuff reverted because you forgot to close that instance as last.

Maybe we should have the option to having just one instance opened? Everytime I double click a model, a second instance is opened. I often rather would open it in the already opened instance.

Maybe the system should differ between an successful change in settings and when something corrupts?

This is the only software I know of where settings get reverted back when having multiple instances open.

Ghostkeeper, 2017-09-25:
> Maybe we should have the option to having just one instance opened? Everytime I double click a model, a second instance is opened. I often rather would open it in the already opened instance.

That option is actually available. You have to start Cura using the `--single-instance` parameter. It's not exposed to the user interface via a preference, but it's used by external macros that start Cura, such as the one for SolidWorks. I'll make a feature request to add this as a preference though.

> This is the only software I know of where settings get reverted back when having multiple instances open.

Nothing gets reverted back. It gets saved by one of the instances. Which of the instances you'd prefer to have saved is unknown to Cura, because it's not always the first instance that was opened. Perhaps we could try `import esp` but I don't think even Python provides that library :wink:

Lots of software saves settings upon closing. We also save settings before closing.

MathijsG, 2017-09-25:
But why not ignore settings that haven't changed? Or make some user-flag if the user manually changes anything, so it won't get overwritten with a previous setting without having it set this way? I find it rather bad usability how it's handled the way it is now.

Ghostkeeper, 2017-10-01:
But saving changes on a per-setting basis would lead to weird results too, wouldn't it? You could be setting the flow to 120% in one instance for stronger wall adhesion, and the infill density to 100% in another, and then get massive overextrusion in the infill next time. Or you could be changing settings for the Fast profile in one instance and switch to Fine in another, and suddenly find that your modified settings have been moved over to the Fine profile.

What if the user doesn't want changes to be made to the profiles and intentionally closes the instance without changes last?

MathijsG, 2017-10-01:
I doubt users even do know of the current weird behavior, and thus will close windows strategically with it.

I'm merely talking about settings that go beyond the whole print profile settings on the side, with that I rather see a disconnection between material settings and speed/print quality material independent settings, but that is another topic.

If we would somehow log which settings are actively made by a user, per setting you could change just one thing, without overwrite everything? I'm more on the user experience side than the code side, but I doubt the current logic with settings implicitly being reverted.

Can't we save upon closing only the actual changes being made instead of just overwrite all settings again?

Ghostkeeper, 2017-10-11:
I think it's good to have a third opinion on this because we're basically just disagreeing on what the expected behaviour is here.

Ghostkeeper, 2018-02-21:
Our project manager removed this from our planning because it's not deemed important enough.