Ultimaker/Cura #2433: Travels when changing extruder/colors

maaspt, 2017-09-14:
I am having  strange travels when using two extruder changing colors with the same hotend.

If the project has two colors but the first layer only has one, the hotend always travel to (x0,y0) no changes on z, on every extruder/color change, no matter if it has prime tower or not. 
If the project has 100+ extruder changes, it goes 100+ times (x0,y0). It afects outer shell finish, and color oozing.

If the objet has two color but they start from first layer, first layer has two colors, I don`t get these travels from X and y to 0, on every extruder/color change.

Are these thavels a default feature, if they are, why ? How to disable them ?

It happens on all objects that do not have two colors in first layer, no matter what extruder or printer configuration I have. 

Best regards

Ghostkeeper, 2017-09-20:
I suspect it's the extruder switch start/end position.

The author of the printer definition you're using should set the setting `machine_extruder_start_pos_abs` to `false`. This way the start/end position gets interpreted as a relative coordinate and moves to relative coordinate 0,0 are left out.

Ghostkeeper, 2017-09-20:
What printer are you using?

maaspt, 2017-09-21:
I have my own build cartasian printers with marlin 1.1.5 , working with  two extruders one nozzle.

I don´t think it is printer definition on marlin, because without any extruder start/end gcode in cura, travels to x0,y0 are visible in Cura, any version.

If you dowload a project from thingiverse with two or more colors, open it in cura merge them, and you can see travels I am talking about.
The project you select, second color can not start in first layer.

Ghostkeeper, 2017-09-26:
I didn't mean a printer definition on Marlin, but a printer definition in Cura.

This is what I get with a proper printer definition in Cura: It moves to the switching bay in the top-right.
[image]

And this is what I get with Custom FDM Printer: In this case, the extruder's start and end positions are 0 and the `machine_extruder_start_pos_abs` is also False, so there are no strange movements to 0 (only the one from the starting location on the first layer):
[image]

maaspt, 2017-10-01:
Select a project that does not start with two color in first layer.

second color starts after first layer.
[image]
[image]

maaspt, 2017-10-01:
please get these project
https://www.thingiverse.com/thing:2524765

Select objects and extruders to start two colors in first layer, I get no travels to (0,0)

If you select objects and colors that you get object with same color in first layer I get travels to (0,0) 

[image]
[image]
[image]
[image]

maaspt, 2017-10-01:
I am on Custom FDM Printer

Already changed machine_extruder_start_pos_abs to false, in fdmextruder.json but getting same travels, maybe I am not changing in the right file.

"machine_extruder_start_pos_abs":
                {
                    "label": "Extruder Start Position Absolute",
                    "description": "Make the extruder starting position absolute rather than relative to the last-known location of the head.",
                    "type": "bool",
                    "default_value": false,
                    "settable_per_mesh": false,
                    "settable_per_extruder": false,
                    "settable_per_meshgroup": false,
                    "settable_globally": false
                },

Ghostkeeper, 2017-10-11:
I'm getting the same result using a similar model that I made:

[image]

I tried modifying the prime position (hence the giant disallowed area that you see there) but it still happened. I suspect it might be a bug in the engine because I get some supposedly empty layers which aren't in the model.

Ghostkeeper, 2017-10-11:
I was able to move half of them by modifying Layer Start X and Layer Start Y. This is because the initial position of a layer is unknown when the travel moves are planned, because the travel moves are planned for all layers at the same time.

Ghostkeeper, 2018-02-21:
Our project manager removed this from our planning because it was not deemed important enough.

epsalmond, 2018-06-30:
@maaspt and others, you can solve this with Extenions > Post Processing script.

Add a find and replace for the negative position moves.  See below.

[image]

TimoErdmann, 2018-09-24:
so any news on this Bug?

Ghostkeeper, 2018-11-21:
Yes, I believe we fixed this in version 3.6 (or was it 3.5?). Could you try again?

xile6, 2018-12-24:
This problem has not been fix in 3.6
I just ran into it today. Really sucks because it rams the hotend into a bed clip.
Any thing to fix this? besides the replace code?

com9, 2019-02-04:
Still unfixed and really dangerous Bug, please, please fix!

paukstelis, 2019-03-09:
Agreed that this is dangerous. This is still in CuraEngine 4.0 (current master).

Ghostkeeper, 2019-03-11:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5383. This issue is older but that issue has some good reproduction steps that will help debug this. Let's move the discussion over there.

closed 2019-03-11 by Ghostkeeper