Ultimaker/Cura #2434: Cura startup performance optimization

Chrisu02, 2017-09-15:
Hello Guy`s

I latley the Cura 2.x Version always took very long on my machine to start, so i was wondering what that would cause.
Therefor i did some time/memory measuring with Cura 2.6.2 on two of my computers:

1.st one is a i5 540@ 2,27 Ghz 8 GB ram / SSD
2.nd one is a i5 6600k@4Ghz  16 GB Ram / SSD

Starting on the 1st Device takes **28.1 seconds** and ram usage is 429 MB
Starting on the 2nd Device takes **11.0 seconds** and Ram usage is 375 MB

I think the different ram usage may come from the differente CPU type, therefor i ignore the difference.

then i startet to delete all not necessary definitions of printer/material/extruder at all three config folders to only have my active Printer with my two default settings loaded.

The loading times and memory usage went dramatically down.

Starting on the 1st Device takes **8,9 seconds** and ram usage is 212 MB
Starting on the 2nd Device takes **3.8 seconds** and Ram usage is 153 MB

So while the memory usage is not that big deal with enough ram, going from nearly an half minute to under 10 Seconds loading time is pretty impressive for only removing unused stuff !

I dont think most user don`t need that :
70 Printer definitions, 
31 Extruders definitions, 
21 Firmwares , 
79 Material definitions, 
30 Print meshes, 
239 Quality definitions
29 Variants definitions
every time they start cura (i don`t know if all of them are loades but it seems at least a lot of them)

Therefor would it be possible to change the boot behavior of Cura ?
Maybe at first to only load the stuff that was active the last time cura was running  ?
Or for filaments to only cache the on used the last X times, load only text entries and load the file when its really used ?

nallath, 2017-09-18:
This is something we need to improve. We're already doing "lazy loading", so even though the profiles are loaded, they aren't completely loaded.

Ghostkeeper, 2017-09-24:
IIRC we only load the metadata. This does require us to parse the cfg file though. But we need the metadata in order to know if the profile is even relevant for the current set-up.

nickthetait, 2017-09-24:
I agree that this is the single biggest headache with the application at this moment.

A ideas to improve this:
* Only load the printer definitions which have been configured
* Defer some of the loading until after the application has launched and user can start doing work
* Is there any way we could cache some of this data?

nallath, 2017-09-26:
The .def.json files are chached already. This is also why the first start is a bit slower (as it's creating said cache).
We need to parse those files as we need an ID + name combo for the add printer combo. We're thinking about fixing a system that handles this a bit more elegant.
The biggest culprit are the materials; A single material can spawn a number of smaller profiles that aren't actually different.

fieldOfView, 2017-09-26:
> A single material can spawn a number of smaller profiles that aren't actually different.

But that could be cached too, right?

nallath, 2017-09-26:
Eeh, sorta. The issue then still is that you create about 400-500 objects. This mostly has an impact on the running speed.

fieldOfView, 2017-09-26:
But surely loading 400-500 objects from a pickle would be immensely faster than parsing xml files and deciding what 400-500 objects need to be created on the fly.

nallath, 2017-09-26:
I'll mess around with a bit to see if it helps. I can vaguely remember trying it before, so i assume there was a reason i didn't commit it.

nallath, 2017-09-26:
Right; So on my system loading everything takes 15.5 seconds. 6.0 seconds is due to all definitions and 7.7 seconds is due to the material loading. This leaves less than two seconds for all other instance containers. So that's a pretty big difference.

nickthetait, 2017-09-29:
These materials we are discussing, are these just for the current printer, all configured printers or all printers that have definitions and materials?

nallath, 2017-09-29:
It loads all materials; A single XML material can spawn a number of profiles; for each configuration it will spawn one.

As for the optimisation; I already managed to knock it down to ~7-9 seconds (from ~15 sec) by optimising the way we cache queries, improving the pickle version that we use and temporarily disable the garbage collection while loading.

These are of course patch jobs; but if we can also improve the actual problem, we might be able to get it down to a few seconds!

nickthetait, 2017-10-01:
Nice work!

ChrisTerBeke, 2017-12-21:
In 3.2 this will be fixed by only loading needed machines and profiles (and lazy-loading newly created ones).

closed 2017-12-21 by ChrisTerBeke