Ultimaker/Cura #2435: CuraEngine mesh_rotation_matrix to rotate 180 degrees around x-axis

wijnandvelzel, 2017-09-15:
Hi,

I am trying to rotate an object before slicing using CuraEngine.

However, if i try to rotate 180 degrees (-s mesh_rotation_matrix="[[1,0,0],[0,-1,0],[0,0,-1]]") slicing fails.

Without rotation slicing works as expected

Snippet of the output:

"

Progress:inset+skin:1:1 0.276473%

Layer count: 1

Progress: inset+skin accomplished in 0.000s

Starting support...

Removing 1 layers because they are empty

Stopping process because there are no non-empty layers.

Progress: support accomplished in 0.000s

Starting export...

;FLAVOR:Marlin

;TIME:6666

;Filament used: 0m

"

Is there anyone who has an idea what I'm doing wrong. Or is this a bug in CuraEngine?

nallath, 2017-09-18:
I'm not quite sure what the status of that parameter is. As far as I know, it's hardly ever used. The command line interface is only designed as a debug interface.

Ghostkeeper, 2017-09-24:
Perhaps the entire model ended up below the build plate due to your rotation?

ianpaschal, 2018-03-09:
Needs more info but has 6+ months with no activity; closing.

closed 2018-03-09 by ianpaschal