Ultimaker/Cura #2437: master branch - all profiles have gone missing today

smartavionics, 2017-09-15:
Hi, sorry to be a pain but I can no longer see my profiles. Something has changed in the last 24 hours that has made them disappear. The profile files still exist under ~/.local/share/cura/master/quality and their setting_version has been changed from 2 to 3. Help!

smartavionics, 2017-09-15:
One other thing, the profiles were present earlier today, I first noticed they had gone after I had loaded a project saved some weeks ago. Could loading the project have some bad side effect? Restarting cura makes no difference, it doesn't show the profiles.

smartavionics, 2017-09-15:
My guess is that it is 84ad9c6ad1edd08216df0834298fd558c1268920 that is causing the problem because if I go back to just before that commit, my profiles are accessible again. Something like that. What do you think @LipuFei ?

nallath, 2017-09-18:
Lipu is on a holiday, so don't expect any answer from him soon ;) 
The whole quality slicer is still going to get a whole lot of polish, so i that this will be resolved soon.

smartavionics, 2017-09-18:
Understood. I hope this gets fixed soon because I can't use the tip of the master branch at the moment.

smartavionics, 2017-09-21:
Tried the master branch again and profiles are now visible once more. Thanks to whoever it was who fixed the problem. Closing...

smartavionics, 2017-09-22:
They have gone again after something was committed recently so reopening this...

Ghostkeeper, 2017-09-28:
Do you have that project file perhaps, so that we can reproduce it? The profiles save and are retained as normal for me.

Maybe a log would help as well.

smartavionics, 2017-09-28:
Hi, we're back to normal again now. I think it was just a transient blip due to the UI development work. Closing...

closed 2017-09-28 by smartavionics