Ultimaker/Cura #2438: Cura 2.7 constantly trying to access floppy drive

Horsj, 2017-09-15:
Windows 10 x64.  I have a floppy drive on this machine for some legacy applications.

When Cura 2.7 is loaded, the floppy drive is constantly being accessed even though there is no disk in it.

nallath, 2017-09-18:
We can't actually distinguish between a SD drive and a floppy drive. As a lot of people use removable drives for their printer, we added a functionality that checks for it and allows for easy saving.

Ghostkeeper, 2017-09-24:
The culprit is here: https://github.com/Ultimaker/Cura/blob/5e0052dcb6e7f7d5d0c66bf5ea51151b425ed8f0/plugins/USBPrinting/USBPrinterOutputDeviceManager.py#L86

We check for USB storage devices every 5 seconds by going down the serial port list. If its hardware ID starts with USB we detect a USB stick. Apparently accessing the serial port gets your floppy disk spinning.

ns78, 2017-10-11:
So how do you stop it

fieldOfView, 2017-10-11:
You could disable the "Removable Drive Output Plugin" on the Plugins pane of preferences and restart

fieldOfView, 2017-10-11:
@Ghostkeeper:
> The culprit is here: USBPrinterOutputDeviceManager.py

Err... no. That lists serial ports, and the floppy drive aint one. Here's the real culprit:
https://github.com/Ultimaker/Cura/blob/master/plugins/RemovableDriveOutputDevice/WindowsRemovableDrivePlugin.py#L56

This asks the question (in a comment) if we really want to skip A & B, but then doesn't skip A & B. And it really should, because A & B are reserved for floppy drives due to legacy reasons.

nallath, 2017-10-11:
Well... Not quite. We have had people that used A & B as SD card / removable drives.

MFHFozzy, 2017-11-04:
I believe fieldOfView is right if the machine in question is running a legacy BIOS (non UEFI).
Windows may allow you to map a drive to A or B, but that is being done in a high layer and being redirected lower down in the filers somewhere....i think.

Could this actually be related too *all* the removable drive problems?

sharpie7, 2018-12-27:
> Here's the real culprit:
> https://github.com/Ultimaker/Cura/blob/master/plugins/RemovableDriveOutputDevice/WindowsRemovableDrivePlugin.py#L56
>
> This asks the question (in a comment) if we really want [...]

For anyone who wants to keep the plugin, but skip A and B then edit the file above (as administrator) to change the line:
`for letter in string.ascii_uppercase:`
to
`for letter in string.ascii_uppercase[2:]:`

commit b34c0e8e (2019-11-04): Don't check A: and B: drives for USB sticks

Fixes #2438.
closed 2019-11-04 by Ghostkeeper