Ultimaker/Cura #2440: Cura should begin printing on the outer side of the object

dr4Ke, 2017-09-16:
I've got a model that have an open outside, so that the brim is printed on the outside and near the center of the model. The issue is that it starts to print the brim at the most inner point, not the most outer. So it prints over the filament extruded during the travel from oozing point to the print start point.
I think the start point should always be at the outer of the model. The closer to the oozing point, the better. What do you think?

I use a third party printer:
[image]

I tried with a Ultimaker 3:
[image]

Notice the first travel trace through the model to the inner side of the brim.

MikeTough, 2017-09-18:
Had the same issue. Brim creation inside to outside caused a 20mm gap at the begining of the travel leaving brim unattached to the part, resulting in failed print. This is worthy of fixing/further investigation.

Ghostkeeper, 2017-09-24:
Even if you print from outside to the inside, there will still be a travel move because it's moving to the Z seam location.

dr4Ke, 2017-09-24:
It's not all travel moves that bothers me. It's the initial one, because it carries unused filament, not attached to the bed. And when it tries to print over this filament, there's a good chance that this filament will be hurt by the nozzle, preventing the brim to attach on the bed.
On the subsequent travel moves, there is only a thin remaining of filament and it's OK.

Ghostkeeper, 2017-10-01:
I'm a bit worried that this will change all of the tensions in the brim. It might be for the better, or might be for the worse. Only some testing will give an answer, I guess.

ChrisTerBeke, 2018-03-02:
There's a setting for this. By default we don't do it because it gives poor printing results.

Ghostkeeper, 2018-03-26:
Which setting is that then? As far as I know there is no setting for where to start with the brim.

hamster65, 2019-01-19:
I'd like to know which setting it is, too. My brim starts at the walls when the nozzle has not been primed. So the brim is not connected to the object, which then warps or comes loose (Cura 3.6.0)

tioteath, 2019-03-26:
This is the real issue on Endre-3. After this initial travel, the head hits this single line of filament each time it goes around printing Brim. This causes visible defect on the print and wears down printer parts.

Ghostkeeper, 2019-03-28:
There is no setting for where to start with the brim. It should take the brim line that is closest to the initial nozzle position first. Usually that would be the outermost brim.

As far as I could reproduce, it's always starting with the outermost brim using default settings.

closed 2020-06-20 by no-response[bot]