Ultimaker/Cura #2448: Single Profile for Multiple Extruders

thesec, 2017-09-18:
I often have the same material and same nozzle size on both extruders of my machines and switch often for e.g. the color only.

It would simplify things to have a mode with a single profile only, so settings do not apply per extruder tab.

Just dropping the idea ... :)

nallath, 2017-09-18:
It's something that we considered, but it's rather hard to get right. We have a feature request for "Copy all settings to other extruder". Would that be something that would fix your problem?

thesec, 2017-09-18:
That'd be nice, yes.

But isn't the 'single profile' just that, only copying with any change?
You could add a checkbox toggle between the tabs to make it obvious when active?

?)

Ghostkeeper, 2017-09-24:
See #2048.

closed 2017-11-14 by ChrisTerBeke