Ultimaker/Cura #2449: Test patch enhancement suggestion

MikeTough, 2017-09-18:
Hi there,

This is my first post here. I'm relatively new to 3D printing, however i do have fair bit of experience in product design/development (Mech Eng degree, PhD, programming, fancy maths, etc.) coupled with 30+ years of precision machine design (including a stint with industrial film extruders).

Cura appears to be a great product, however finding the appropriate "sweet-spot" for some of the settings I find to be a tedious and time consuming process :(

It would be great if Cura could have some built-in test patterns which could allow the impact of various settings to be explored in a single print. Perhaps this is best illustrated by example:

Getting good first layer adhesion on some filaments can be tricky, and Cura offers a number of settings to improve this, including, nozzle temp, extrusion speed, Z axis offset, initial layer thickness, initial layer width, etc. etc. Exploring these options using individual test piece runs takes ages.

If Cura could print a matrix of, say 25mm x 25mm single layer test pieces (say 5 x 5) where each piece could have a different Cura setting, then optimisation (or at least a good starting point for further tweaking) would be much faster. On the 5 x 5 matrix you could run a range of extruder temps on one axis and extrusion speed on the other. Ideally, one would set the "static" Cura settings for the test, then select two variable settings (one for rows and one for columns) and provide a list or range of values to be printed.

It's easy to see how this could be extended to different test patters to explore different issues. For example, retraction settings could be a matrix of stop/gap/start lines each with different retraction settings. Bridging, brim settings, bottom skin settings, etc, etc.

I'd give this a go myself, but I'm afraid, for me, 3D printing is a means to a end - producing prototype parts for my project. 

So, interested to hear if anybody else out there thinks there is merit in this and if anybody would be prepared to pick up the challenge :)

nallath, 2017-09-18:
This is why at Ultimaker, we have a pretty big team who's sole responsibility it is to come up with the print profiles.

Ghostkeeper, 2017-09-24:
There have been plans to guide people through the process of creating profiles for their materials from within Cura. Cura would suggest tweaking a number of parameters focussed on the specific parts of the print, based on real-world observations of print quality.

Start with the temperature settings, then cooling, that's all that I know.

nickthetait, 2017-09-24:
Mike, welcome to github and 3d printing :smile:  For your own purposes, here are a few of the most commonly used test print models:
* 3D Benchy http://www.3dbenchy.com/download/
* Make: shootout models https://www.thingiverse.com/thing:2071696/#files
* my personal favorite quick test, 3D Hub's Marvin https://www.youmagine.com/designs/3d-hubs-marvin-key-chain--4

Now, back to the idea itself. I agree that a "setting tester array" would be a desired feature. Here is an example request that I remember from this week... https://twitter.com/3DPProfessor/status/910913146289717248

On a technical level, now that it is possible to customize settings per model, I think this feature is within reach. Let's do some brainstorming on how it would work...

@Ghostkeeper's idea centers on a suggestion system on what settings to figure out and in what order. But I think the more desired idea is being able to test several copies of a model in a single print job. A system for being able to identify which settings were used on each model is probably the most important missing piece right now. Using a [model-etching technique](https://www.academia.edu/11229348/Polymer_recycling_codes_for_distributed_manufacturing_with_3-D_printers) would be one option. Another is 3dprof's method to print text next to the relevant model, but this is separate from the model and can be lost.

@nallath if you have a minute, ask the materials team how much they would pay for a feature like this :stuck_out_tongue_winking_eye:

ianpaschal, 2018-03-05:
I'm going to go ahead and close this due to age and because the materials team does use this sort of workflow already. Plugin authors are welcome to make something available to the community though.

closed 2018-03-05 by ianpaschal