Ultimaker/Cura #2451: Cura will not show helpers/layers

Racingf3, 2017-09-18:
After upgrading to Cura 2.7 i can not see layers. I have Cuta also on my laptop (2.7) here it works fine.

smartavionics, 2017-09-18:
Sounds like a graphics driver problem, are you using the latest OS/graphics drivers/etc?

Racingf3, 2017-09-18:
Yes, all updated.

I can se travelers.

Thor-Christian Ebbesvik
Daglig leder
Autosalg A/S
tc@autosalg.no<mailto:tc@autosalg.no>
www.autosalg.no<http://www.autosalg.no>

Ghostkeeper, 2017-09-24:
Strange. Just to be sure, you did activate the check marks for the Helpers, Shell and Infill in the layer view panel, didn't you? 😛 

You could try upgrading your graphics drivers. The intended rendering requires OpenGL 4.1.

Or you could set Cura's rendering to legacy mode: Go to Preferences -> Configure Cura... -> Force layer view compatibility mode (restart required). This will disable the fancy 3D tubes in Layer View, but at the paths should become visible again. The compatibility mode only requires OpenGL 2.0.

Racingf3, 2017-09-25:
I am running gtx780 ti so should work fine.

Works on Cura 2.6

Thor-Christian Ebbesvik
Daglig leder
Autosalg A/S
tc@autosalg.no<mailto:tc@autosalg.no>
www.autosalg.no<http://www.autosalg.no>

ianpaschal, 2018-02-27:
@Racingf3 Is this still an issue for you in the latest version (3.2.1)? Since it's possibly graphics card specific I can't test it myself before closing.

closed 2018-03-05 by ianpaschal