Ultimaker/Cura #2453: Layer View mode doesn't show up

oaspereira, 2017-09-19:
Hi all!

I'm having problems in this last versions of Cura 2.5/2.7 with layer view mode.

It's simply gone!!!

[image]

[image]

Already tried to force layer view compatibility mode on Preferences and restart without results
[image]

What to do??

nallath, 2017-09-19:
Could you share the logs with us?

oaspereira, 2017-09-19:
Hey I solved it!
Uninstall Cura and deleted APP DATA folders
Installed again

It's working now

Lost all configurations but it's worth it

Thanks anyway

closed 2017-09-19 by oaspereira

biscuitlad, 2018-01-31:
I've got the same issue. It seems to happen when you try to upgrade to the latest version. If for some reason you need to revert (hello, stable versions?!), then sadly the layers view disappears. Like just vanishes without trace...

`018-01-31 15:46:27,315 - ERROR - UM.Qt.Bindings.PluginsModel._update [40]: Plugin is missing a plugin metadata entry
2018-01-31 15:46:27,315 - ERROR - UM.Qt.Bindings.PluginsModel._update [40]: Plugin is missing a plugin metadata entry
2018-01-31 15:46:27,315 - ERROR - UM.Qt.Bindings.PluginsModel._update [40]: Plugin is missing a plugin metadata entry
2018-01-31 15:46:27,321 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting info/automatic_update_check.
2018-01-31 15:46:27,324 - WARNING - UM.Preferences.getValue [81]: Tried to get the value of non-existing setting view/force_layer_view_compatibility_mode.
2018-01-31 15:46:37,544 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles`