Ultimaker/Cura #2454: Cura 2.7 mouse zoom resolution - Reversed

Thisismydigitalself, 2017-09-19:
It seems zooming in/out RESOLUTION (not to be confused with DIRECTION) in Cura 2.7 by mouse wheel looks to be reversed, e.g - when mouse is set for low line count scroll, Cura zooms very aggressively and vise versa. 
 
Steps to Reproduce
Change mouse setting to scroll less lines - Cura zoom resolution is more coarse, and vise versa
 
Actual Results
When mouse scroll wheel set to low line count,  Cura zooms in/out very abruptly and vise verse

Expected results
Should be similar to Windows scroll behavior - less lines - higher resolution  / smaller steps movement.

Ghostkeeper, 2017-09-24:
For me on Windows 8.1, the zoom speed doesn't seem to be affected at all by the scroll speed. When I've set it to scroll 3 lines at once (my default) the zoom speed seems to be the same as when I set it to scroll 1 line at a time.

Thisismydigitalself, 2017-09-24:
Thank you for your insight. following your findings I double checked and found the culprit. I am using Logitech Trackball mouse and every time i open its control app and touch something it messes up the mouse wheel resolution in Cura. when i then open Win10 mouse settings and touch the scroll lines it resets somehow and as you said line count does not have influence on zoom resolution anymore.

ianpaschal, 2018-02-27:
@Thisismydigitalself Am I reading correctly that you found the problem outside of Cura and managed to get it worked out?

Thisismydigitalself, 2018-02-27:
This is an old one :).
I don't have any issues with zoom in/out anymore.
Case closed.

closed 2018-02-27 by ianpaschal