Ultimaker/Cura #2455: Cura 2.7: (-)Z height from print bed

Thisismydigitalself, 2017-09-19:
When opening a project with model which is -Z height from printbed and that is while general settings is set to drop model down to built plate, the latter overrides the saved project -Z value and reset it. This might be on purpose but I wish to have a message box telling me that this is the case and whether I want to go with my file settings or let Cura drop in to print bed. 
 
**Steps to Reproduce**
Set general settings to "Drop model to build plate"
Save project with -Z value for height.

**Actual Results**
Cura will disregard the -Z value and reset it to Z=0

**Expected results**
Warn me before doing so so i can chose whether to go with my saved settings or let Cura drop it to built plate. if Cura makes a change and overrides a file configuration, it must warn the user.

I have noticed this by trying to slice not so perfect STL's from the internet where there are some imperfections on bottom surface and I was hoping  that by dropping these 0.1 or 0.2 down to built plate seems to be doing the job (i get flat bright red surface looking from the bottom). I know this is not "legal" but a way to get rid of small surface imperfections.

ChrisTerBeke, 2017-09-19:
A fix for this is already underway!

Details: https://github.com/Ultimaker/Cura/commit/e18b24add294557c726fce637294a9d790d606d5

Thisismydigitalself, 2017-09-19:
Cool!

Ghostkeeper, 2017-09-24:
Fixed in 3.0.

closed 2017-09-24 by Ghostkeeper