Ultimaker/Cura #2456: Cura 2.7: Reset Viewport in zoom in "Zoom / center to mouse location".

Thisismydigitalself, 2017-09-19:
Due to mouse resolution issue the viewport went crazy and the built plate got lost from screen. i could not get it back to view other than to kill and restart Cura.

**Steps to Reproduce**
Set Zoom to mouse location  - checked, and play with zoom till Built plate disappears.
 
**Actual Results**
A way to reset view - i might missed this option . could not find it in GUI nor in right click menu.

**Expected results**
Reset view button.

fieldOfView, 2017-09-19:
See https://github.com/Ultimaker/Cura/pull/2439

Thisismydigitalself, 2017-09-19:
Cool!

fieldOfView, 2017-09-19:
This can be closed now that #2439 is merged

nallath, 2017-09-19:
It was a bit too late for next release (it has a string, so it will only be in the release after the next). Sorry about that.

closed 2017-09-19 by nallath

CoreyBurger, 2022-12-15:
Can we reopen this? I've ended up with a view I cannot reset and the "Reset View" option in #2439 seems to have vanished in 5.1.1