Ultimaker/Cura #2457: SidebarSimple sliders are not DPI aware on Windows

fieldOfView, 2017-09-19:
The new sliders on the Recommended tab render inconsistent on scaled displays (eg Hi DPI screens):
[image]

* The layer height texts are much too small on the top slider
* The marker dots are much smaller than on 100% scaled displays on the top slider
* The bottom slider is not wide enough
* There's an inconsistency between the linewidts of the two sliders

Steps to reproduce:
In "Display settings" (context menu on the desktop), set the "Change the size of text, apps and other items" setting to anything but 100%
Restart Cura

Other things to note: splash-screen alignment is broken as well, and many dialogs and margins in dialogs are too small.

nallath, 2017-09-19:
I've added your remarks to the issue regarding the slider.

Appesteijn, 2017-09-25:
This should be fixed with the latest changes to the sidebar

ChrisTerBeke, 2017-09-27:
Fixed by today's changes to DPI calculation.

closed 2017-09-27 by ChrisTerBeke