Ultimaker/Cura #2458: Cura 2.7 "layer start" issue (concerning Z-seam)

ozfunghi, 2017-09-19:
When printing one single object in 2.7, and you move the Z-seam to the back, the "layer start" coördinates will not follow the Z-seam, making a seam where the layer start coördinates are. I contacted people at Cura that said this is not supposed to happen when printing a single object (as layer start is supposedly only for multiple objects) but the proof is there. See the screenshot files from Cura, that are provided. Same printing setup with only one difference, the layer start coördinates. See the picture of the two prints, both with Z-seam at the back, but one with layer start at default (creating a nasty seam at the front) and one with layer start at the same coördinates as the Z-seam, at the back. 

The inner walls will create a seam which can not be covered up by the outer wall. Only the outer wall starts where the Z-seam is set. In this case, you will get TWO seams, one for Z-seam coördinates, and one for layer start coördinates, defeating the purpose of using the Z-seam feature.

Extra negative effect, is that the needless travels, take up 11 estimated minutes longer to print, since the printer needs to travel between Z-seam and layer start coördinates for no good reason.

If this is a bug, it needs to be addressed, if it is not a bug, i would make it more clear, that people who want to have the Z-seam at a certain location, they also have to change the layer start settings.

[image]
[image]
[image]

smartavionics, 2017-09-19:
Hello. Yes, this is a known problem (well, at least to me!) A recent change to Cura has made the inner walls start at a different position to the outer wall and so you get the extra travels and the seam on the inner wall not being where you want.
But all is not lost, the next version of Cura should have an option to optimize the order in which the walls are printed and a side effect of the optimization is that when the z-seam location is specified for the outer wall, the seams of the inner walls will be positioned as close as possible to the outer wall seam.

ozfunghi, 2017-09-19:
Oh, ok. I think you can already chose the order (inner/outer wall). But for the moment, it's good for others to know they can get around the problem by putting the coördinates of layer start and Z-seam in the same location.

Ghostkeeper, 2017-09-24:
I've been investigating a bit.

Normally, Layer Start should only be used for ordering parts: which polygon gets printed first. But it seems that the mechanism in which the way in which the Layer Start influences the print is via the `getLastPlannedPositionOrStartingPosition()` function of `LayerPlan`, which is used everywhere. It takes the last known position or, failing that, the starting position of a layer (per extruder).

When adding the inner walls, we add them with `addPolygonsByOptimizer`, which calls `getLastPlannedPositionOrStartingPosition`. That case is somewhat easy to fix up, since if the z seam is user-specified we have to use that position instead of the last planned position.

However there is also the `moveInsideCombBoundary` function which is called at every layer part, which move slightly inwards to prevent retracting on the wall. If no move has been planned yet on this layer, it'll move inwards from the starting position of the layer, causing a move towards layer_start_x and layer_start_y. Though you'd think that this only happens when there is already stuff planned on a layer (such as the walls that it's supposed to move away from)...

jackha, 2018-07-12:
Is this still a thing? There has been many changes in the order in which perimeters are printed.

jackha, 2018-08-16:
I've quickly tested custom z seam and it works in the current Cura version.

closed 2018-08-16 by jackha