Ultimaker/Cura #2459: Cura 2.7 extremely slow movement on new layer start positioning

MarioStoilov, 2017-09-19:
I'm using the latest cura (2.7.0) non beta. I was messing around with some line width settings and after that, whenever the printer should start a new layer, its movement to the start position of that layer is extremely slow (I would say its crawling). Reverting the settings I changed did not help.

Here is part of a GCODE that thisappeared in (right at the layer change)

`G1 X99.735 Y110.766
M204 S5000
G0 F4500 X97.97 Y93.47
G0 X97.75 Y93.25
;TIME_ELAPSED:60.721651
;LAYER:1
M140 S55
G1 F3000 E128.65762
M106 S255
G0 F6750 X97.75 Y93.25 Z0.45
M204 S500
;TYPE:WALL-INNER
G1 F3000 E135.15762
G1 F3150 X116.25 Y93.25 E135.49595`

smartavionics, 2017-09-19:
This is a known bug. It's all to do with the z-hop feedrate getting used for subsequent travel moves. Do you have retract on layer change enabled? If so, try turning that off. If that doesn't help, turn off z-hop. I have provided a fix for it and suggested that they issue a 2.7.1 but I don't think that is going to happen.

MarioStoilov, 2017-09-19:
Yup, that fixed the issue. Will leave this open (it is still a bug)

devils27, 2017-09-19:
I see the post with the your code difference but how do I use the fix?  Is the only way to include that fix to try and compile and install CuraEngine and use that?  Or is it possible to use it with my Cura 2.7 on windows by modifying some code?  Thanks in advance and sorry if this a newbie question or the wrong place

smartavionics, 2017-09-20:
Hi. Yes, I'm afraid the only way to use the fix is to compile the cura code yourself. To do that you have to get a bunch of prerequisite libraries and it's really quite an effort.

krisrok, 2017-09-23:
i have the exact same issue: all movement is fine until layer change.

interestingly, i have z-hop enabled but "retract at layer change" is disabled. so in my case printing without it does not fix it like @smartavionics suggested.

smartavionics, 2017-09-23:
Hello @krisrok, please test the next beta release of cura and report back whether the problem has been fixed for you or not, thanks.

krisrok, 2017-09-25:
will do. so that'd be 2.8 beta?

smartavionics, 2017-09-25:
I think it's actually going to be 3.0 (sounds grand, don't know what new goodies it will have apart from my own contributions).

ianpaschal, 2018-03-05:
Since we didn't hear back from @krisrok, and are now at version 3.2.1, I'm going to assume this has been fixed.

closed 2018-03-05 by ianpaschal

krisrok, 2018-03-05:
yep, that's correct

rholt51, 2018-04-10:
So - all of a sudden (over the past week), i'm having this same problem. (Cura 3.2.1) It's taking forever for the head to travel from one object to another!!!! I'm using Marlin which was installed about 6 months ago.
Russ from Coral Springs, Florida, USA

smartavionics, 2018-04-10:
Hi @rholt51 can you please post the gcode you are having a problem with.