Ultimaker/Cura #2460: Suggestion: rename and move "hollow out objects" feature or replace with "internal support"

ozfunghi, 2017-09-19:
When printing with 0 infill, you can have overhangs and bridges that need to be supported internally, on the inside of the print. I've always been looking for a way to place "internal" supports, just for when i print without infill. Appearantly, this is possible with the "hollow out objects" feature (although when chosing this in combination with "supports everywhere" i still don't see any internal supports in my Cura preview unless i turn my infill on, so i'm not even sure the feature works as advertized in 2.7 and it kind of defeats the purpose if you need infill anyway). But both name and placement make no sense, imo.

The name suggests it makes the print hollow, as in "without infill" (yet it doesn't for me in 2.7 as i pointed out above).  But why would you go all the way down to look up this feature called "hollow out objects" under experimental, when you can just turn infill to 0% to make them hollow. On the other hand, if you are looking to place supports on the inside of the print, you will be looking under the "support" tab or a name that reflects what this does exactly. Furthermore, the only reason to use this feature (and not just use 0% infill) is to be able to place internal supports, because again, if you want to make it hollow, everybody will just set infill to 0, so there is no reason to be so unclear with the name to begin with.

So i suggest some modifications:
Rename and move it (or replace the feature completely) so that there is an extra feature under "support" and name it "internal supports". Furthermore, it should be possible to select internal and external supports individually. I may not want supports on the outside of my print (for cosmetical reasons) but i may want supports on the inside of the print (for both structural and cosmetical reasons). Alternatively, add an option under infill, to place infill only where it can serve as support (with overhang value).

If people really want a "one click solution" to hollow out a model (regardless of support structure), it would make sense to place this under "infill". Select it, and the infill box will go gray, making for 0% infill. Deselect it and you can change the percentage of infill required. That way you can easily switch between 0% and a default % which you like to use. Not unlike selecting supports, where the overhang angle and other settings are remembered by Cura when turned on.

An internal supports feature, under the support tab would be self eplanatory, it could be easily found (under support), and if turning it on, should at the same time remove the infill, it can easily be explained in a popup message, what the consequences are, just like it is now with every other feature. 

In short: 

Remove the "hollow out objects" from "experimental".
Add an "internal supports" option under supports. 
Put the current support options under "external supports" and place that under "support"
Like this;

SUPPORT
-external support: no/yes + settings
-internal support: no/yes + settings

Optionally add an option "make hollow" under infill, to automatically set infill to 0%. Or alternatively place this option under the "internal supports" feature.

Alternatively, add an option under infill, to place infill only where it can serve as support (overhang value), so that it is not needed to print an entire model with infill, if it is only needed in areas where it serves as support.

Ghostkeeper, 2017-09-25:
> i still don't see any internal supports in my Cura preview unless i turn my infill on

Yes, this seems to be a bug. Even when you set the infill density to 0%, I'd expect Hollow Out Objects to work. Reproduces for me as well.

> Furthermore, it should be possible to select internal and external supports individually.

This is a problem with our current architecture. We are unable to distinguish between support that is inside a hollowed out object or support that's on the outside, due to the nature in which Hollow Out Objects is implemented: The infill areas are subtracted from the model itself before support is generated.

I agree that a one-click infill setting for support-like infill would be much better. All of the infill settings like gradual infill, infill density and such would still work. It's just kind of hard to implement as such.

Ghostkeeper, 2017-12-11:
This issue was rejected by our project manager because it's too much of an edge case.