Ultimaker/Cura #2461: Wrong fan speed for first raft layer

JPFrancoia, 2017-09-19:
Hi,

I'm using cura 2.7.0-1 (installed from Archlinux repositories).

And here are my settings:

[image]

As you can see, my "Raft fan speed" is set to 0.

Here is the gcode generated from this config:

[test.gcode.txt](https://github.com/Ultimaker/Cura/files/1315221/test.gcode.txt)

Here is the interesting part in the Gcode:

```
;LAYER:-5
M106 S255
M204 S625
M205 X6 Y6
```

This is the first layer of the print (first raft layer). The fan is ON at full power.

```
;LAYER:-4
M107
```

And the fan is turned off at the second raft layer. I think it's a bug. Can you reproduce it ?

Ghostkeeper, 2017-09-25:
Hmm, no I can't. I did two experiments:

Regular fan speed to 50%, regular fan speed at height to 0mm, enable raft. This should reproduce your settings regarding fan speed. But all I saw regarding `M106` was:

    ;LAYER:0
    M106 S127.5

This is at layer 0, so after the raft. At subsequent layers there were more `M106` commands because of the minimum layer time.

In the second experiment, I also set the raft base fan speed to 30%, raft middle fan speed to 20% and raft top fan speed to 10%. These manifested itself with additional `M106` commands at the correct locations in the raft. So everything seems to be working all right on this end.

Perhaps you can share a project file?

JPFrancoia, 2017-09-27:
I can share my profile if you want:

[profile.curaprofile.zip](https://github.com/Ultimaker/Cura/files/1337075/profile.curaprofile.zip)

Could you please try it and tell me if it changed something on your side ?

Ghostkeeper, 2017-10-01:
I'm getting different but similar results as you, using that profile.
I set the printer to UM2+, nozzle to 0.6mm and material to PP. This happened at layer -5 (the raft base layer):

    ;LAYER:-5
    M106 S127.5
    M204 S625
    M205 X6 Y6

So instead of S255 (like you) or 0% it would set the fan speed to S127.5 or 50%. At the next layer it turns the fan off:

    ;LAYER:-4
    M107
    G0 X86.455 Y96.354 Z0.68

Then at layer 0 it turns the fan back on at 50% and continues as normal.

This seems like a bug to me, but we'll have to narrow it down a bit further to see what is actually causing this.

JPFrancoia, 2017-10-03:
Ah yeah, if it's 50% I know why. I think I exported my production profile, not the test one. The screenshot in the first message is the test profile, and you can see in the cooling part that maximum fan speed = 100%.

In the production profile, maximum fan speed is set to 50%.

In short: at first layer of raft (-5), the fan speed is set to maximum fan speed. At layer -4, the fan is disabled.

jackha, 2018-07-12:
Is this still a thing in Cura 3.4.1? I would like to close it.

JPFrancoia, 2018-07-19:
No idea, I still use the 3.2 because each time I upgrade Cura it breaks compatibility and I can't afford that at work. I'll close it for now and I'll re-open it if it's still there when I upgrade.

Thanks

closed 2018-07-19 by JPFrancoia