Ultimaker/Cura #2462: Cura 2.7 font issue on Print Setup panel

dinowchang, 2017-09-20:
The first 3 rows (lines) of Print Setup panel, "Quality", "Layer Height", and "Initial Layer Height" shown is abnormal (kind of squeezed).
After scroll down a little bit, the font will become normal.
My system is Win10, and the resolution of display is 1920x1200.

[image]

[image]

Ghostkeeper, 2017-09-25:
It seems the text is not aligned to pixels.

I'm not able to reproduce this bug here though. Maybe it's all different because I'm on Linux and have a different screen size or DPI.

waspinator, 2017-09-25:
same issue on macOS Sierra. 

[image]

ianpaschal, 2018-02-27:
This will be fixed in the next release of Cura (3.3).

closed 2018-02-27 by ianpaschal

Ghostkeeper, 2018-03-20:
The original issue should already be fixed for 3.0 for most places where it occurred.

dinowchang, 2018-03-31:
It is fixed in cura 3.3beta