Ultimaker/Cura #2463: Cura 2.7.0 crashes with Bus Error when building layer view

rosswille, 2017-09-20:
This bug occurs while running Cura-2.7.0.AppImage 64-bit on Ubuntu 16.04.3 LTS.

The following steps reproduce this bug 100% of the time for me...
1. Select a printer with a build volume of 200x200x300mm or greater.
2. Open EiffelTower_fixed.stl from https://www.thingiverse.com/thing:912478
3. Click on the object
3. Scale to 200%
4. Change "View Mode" to "Layers"
5. After the "Processing Layers" progress bar reaches 100% Cura crashes with a "3935 Bus error" message.

The problem does not occur when I skip the scaling step (#3).

smartavionics, 2017-09-21:
Hi. I'm running a version built from recent source rather than using the appimage and it can slice and display the layers OK on my system (same OS as you). Maybe by scaling up the model you are exceeding some capability in the graphics system. Does /var/log/Xorg.0.log or any other system log file contain any hint of the problem?

nallath, 2017-09-21:
Could you share your logs?

awhiemstra, 2017-09-21:
Note that I have seen the occassional bus error on my machine, this usually means you are running with the Intel driver and your video card runs out of memory. Unfortunately there is very little we can do about it, since it is primarily a driver bug.

rosswille, 2017-09-21:
I have uploaded the log file from .local/share/cura/2.7/cura.log.  I'm fairly new to Cura... are there other log files you'd like to see?
[cura.log.zip](https://github.com/Ultimaker/Cura/files/1321954/cura.log.zip)

I'm running on a i5-6600K processor with 16GB RAM and a 3840x2160 display.
When I tail -f /var/log/Xorg.0.log while causing the crash nothing gets posted to the Xorg log file.

smartavionics, 2017-09-21:
I'm using an i3 with 16GB of ram with 1920x1090 screen (perhaps you have too many pixels?)!

rosswille, 2017-09-21:
I will try this on a laptop computer with less pixels, but it happens even when the Cura window is much smaller than the display size. @smartavionics I've never heard of anyone complaining about too many pixels!

rosswille, 2017-09-21:
Ok, I tried the same procedure on a laptop with an i3-2365M CPU with 8 GB RAM, running Ubuntu 16.04.3 and it worked!  It was taking **forever** to slice, so I noticed that Generate Support was enabled and turned that off.  Eventually it finished slicing and then completed the Processing Layers and displayed the layers.

The slicing/processing-layers steps take a lot longer on the i3 compared to the i5, but it doesn't crash.  So, it looks like it's probably a graphics or driver problem rather than a program bug.

One more data point: I get the same Bus Error on two different i5 computers, so the problem isn't flaky hardware.  Maybe, just maybe, this problem will get resolved someday...

Ghostkeeper, 2017-09-26:
I don't think it's necessarily a driver bug either, but just that the GPU is running out of memory. Not the CPU but the GPU. Nothing the drivers can really do about that other than give a Bus Error.

Advanced graphical software, such as AAA games, try to request or otherwise probe the amount of memory available on the GPU and try to fit the game in that memory, reducing graphics quality or sacrificing frame rate as needed. That's a bit out of scope for Cura though. Perhaps in the distant future...

closed 2018-03-02 by ChrisTerBeke