Ultimaker/Cura #2464: Deactivate Extruder / Hotbed

thesec, 2017-09-20:
In a multi extruder setup I do not always, actually rarely, print with all two.

I
Right now I am editing the gcode manually to prevent the unused extruder from heating up.
The other option would be to set all temperatures manually to 0 for that one?
Would be nice to get an option or have Cura auto-detect nothing being printed with an extruder and not generating the gcodes to heat these up.

II
With the extruders printing on the same hotbed ... how does Cura determine the hotbed temperature if multiple are active but with different hotbed temperatures?
Wouldn't it make sense to have the hotbed settings only once in the UI, separate from the extruders?

nallath, 2017-09-21:
This is probably because your start-gcode needs a start temperature. For the UM3 we solved this by making the machine smarter; No start-gcode is required, the machine selects this from the header.

As for number two; We have a "resolve" mechanism. Say that you are printing with PLA and ABS and that they use 50 and 80 for bed temp respectively. As the bed temp can only be set once, cura will look at both values, work a bit of magic (in this case use the maximum value) and use that.

thesec, 2017-09-22:
Thanks for the explanation.

On I - what is the "start-gcode"? Since the machine works fine when the gcodes to fire up the second nozzle are being removed I doubt it needs one in specific. In general an option would help to save some energy, at least :)

On II - can we get that bit of magic **alternatively** to become a rational decision between two ears and have the user decide? ;)

nallath, 2017-09-26:
The user can always decide; If you override the bed value that value is used. 

Machines can have start & end-gcode and extruder switch g-code. Its a set of instructions that always get exectued at certain stages (eg; start of print, end of print, when swithcing extruders). You can change them in the custom machine action.

Ghostkeeper, 2017-09-26:
On I: This is a duplicate of https://github.com/Ultimaker/Cura/issues/1128 . In short, we do detect the nozzle being used, but there still seem to be some bugs in that. Nallath suggested it might be a problem with the start g-code in your case. To see the start g-code, click on Settings -> Printer -> Manage Printers... and then click on "Machine Settings". There's a field called "Start GCode" there, where there's a piece of g-code that gets executed at the start of the print. If that g-code sets any temperatures, the nozzle might be inadvertently heated up.

On II: Nevermind about that resolve mechanism. There is only one bed temperature. It's a global setting. Like all other global settings, you can change the setting in the interface but it's not specific to one extruder. So if you change it in extruder 1, it'll still be changed when you switch to extruder 2. To the average Joe, it's just a rational decision between two ears and the user decides.

The resolve mechanism comes into play when there's two built-in profiles for both materials in either extruder, and they each specify a different build plate temperature. In that case, we choose the highest temperature of the two.

ChrisTerBeke, 2018-03-02:
Planned for upcoming release.

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-26:
It is now also actually implemented: https://github.com/Ultimaker/Cura/pull/3436

thesec, 2018-03-26:
Kudos!