Ultimaker/Cura #2465: Cura 2.7 suggestion: Visualizing nozzle X/Y movement frequency

Thisismydigitalself, 2017-09-21:
Sometimes it is hard to tell where during the print the Gcode / Print head would rock the machine violently, especially with heavy print beds let alone with heavy model growing on these. Would it be wise to visualize this somehow on layer view?

Just thinking out load.

nallath, 2017-09-21:
Well, if this is a problem, i think it would be best to change your acceleration settings.

Thisismydigitalself, 2017-09-21:
Let me try and explain my point better:
It's not about reducing the overall acceleration which might be a bad chose for large prints due to increase in print time. what i was hoping was to know in advance where the nozzle goes back and forth in a small narrow part of the print when high acceleration setting can and will shake the printer.
I would love having dynamic acceleration and a good visualization of where in the print it will take place.

Very nice to have.
Hope this feature will end up on this already very long to-do list.

Ghostkeeper, 2017-09-26:
I think that visualising the speed is more important than visualising the acceleration...

We do calculate the acceleration and jerk for the time estimates, but this is a ["simple" mathematical operation](https://github.com/Ultimaker/CuraEngine/blob/896d6fbadad004f13049a7ebf7dbc90c03e9873a/src/timeEstimate.cpp#L130). To visualise that we'd need the engine to send 2 to 3 times as many lines to the front-end and an additional acceleration field for each. I say 3 times because there's one part of each line where it is accelerating, one part of some lines where it has a constant speed, and one part of each line where it is decelerating. We could do with less: We could let the engine send a fraction-of-the-line-where-it-is-accelerating sort of fractional value. This would be easier to send over.

However all that is very little indication of the actual vibrations going on in your printer because of jerk, which is instantaneous in a corner. To visualise jerk you'd have to visualise the vertices somehow...

Thisismydigitalself, 2017-09-30:
It is more about visualizing frequency rather than speed.
Given Speed to be a global setting and fixed, we need to spot before printing where the nozzle will vibrate harder due to fast change in direction - mostly in tight / narrow stitches where the print-head freaks out. 

Low Frequency:
[image]

High Frequency:
[image]

Ghostkeeper, 2017-10-05:
Yeah, I got what you meant. I interpreted that as that you want to be able to visualise acceleration.

abdavid, 2018-02-12:
I want to bump this and add my two cent here. 

It would be really neat if Cura would support visualising the print progression. Cura already has support for simulating each layer before printing - and interfacing with the printer to find out which line is currently being drawn and mapping that to the emulation does not sound like a whole lot of work since both of the datasets are available. Of course there would be some latency, and I am not really expecting Cura to be able to mirror printer movement realtime, but "ish" would be a nice addition. 

Possibly low hanging fruit? Im not sure as I do not really know the internals of the emulation nor a lot of gcode just that Pronterface already does this, but their app is lacking a whole lot of other functionality which I appreciate in the Cura app.

fieldOfView, 2018-02-12:
Other than eyecandy, how will this benefit users?

abdavid, 2018-02-12:
> Other than eyecandy, how will this benefit users?

Got me there - it would mostly be eyecandy I will give you that. Though I would choose that progression visualisation over the progress bar any day. I must add that I really love the emulation support. It is a nice way to learn what consequences different settings has for the print in a simple and elegant manor.

Ghostkeeper, 2018-03-07:
Unfortunately there is no g-code command for getting the current line number as far as I know. It's impossible to know at what line the printer currently is for most printers.

Another problem is that the visualisation in Cura after slicing is not generated from g-code but generated separately by CuraEngine for performance reasons.

Thisismydigitalself, 2018-03-07:
You guys know better than me.
I was hoping for something like feedRate/Layer Thickness maps but if this is not easy to achieve than this suggestion can be closed for now. thanks for considering.

Ghostkeeper, 2018-03-30:
I think visualising the acceleration would come pretty close to what an expert needs to know to determine if vibrations are going to be a problem. Getting vibration speeds and frequencies is unfeasible, so let's merge this together with https://github.com/Ultimaker/Cura/issues/3186, which has a similar goal.

closed 2018-03-30 by Ghostkeeper