Ultimaker/Cura #2466: Fatal Crash Report V2.6

ItzFitz2113, 2017-09-21:
I have a Taz 6 and recently downloaded the beta version of lulzbot edition 2.6.21. When i tried to slice a model of baby groot and view it in layer view, it immediately crashed.

Here is the Crash Report:

Version: 2.6.21
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 184, in _render
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 138, in render
  File "C:\Program Files (x86)\Cura2 2.6\plugins\LayerView\LayerPass.py", line 95, in render
    batch.render(self._scene.getActiveCamera())
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 215, in render
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 246, in _renderItem
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGL.py", line 199, in createVertexBuffer
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 224, in getColorsAsByteArray
MemoryError

-------------------------
Thread 0x00001510 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00000dc8 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00001170 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00001d90 (most recent call first):
  File "C:\Python352\lib\threading.py", line 293 in wait
  File "C:\Python352\lib\threading.py", line 424 in acquire
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00001040 (most recent call first):
  File "C:\Program Files (x86)\Cura2 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "C:\Python352\lib\threading.py", line 862 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00002038 (most recent call first):
  File "C:\Python352\lib\threading.py", line 297 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1569 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1153 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x000000cc (most recent call first):
  File "C:\Python352\lib\threading.py", line 297 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1569 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1017 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00000c3c (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 946 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00000420 (most recent call first):
  File "C:\Program Files (x86)\Cura2 2.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "C:\Python352\lib\threading.py", line 862 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x000014b0 (most recent call first):
  File "C:\Python352\lib\threading.py", line 297 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1569 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1153 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00002594 (most recent call first):
  File "C:\Python352\lib\threading.py", line 297 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1569 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1017 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00000bc4 (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 946 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00002860 (most recent call first):
  File "C:\Python352\lib\threading.py", line 297 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1569 in wait
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 1017 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Thread 0x00002dac (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\zeroconf.py", line 946 in run
  File "C:\Python352\lib\threading.py", line 914 in _bootstrap_inner
  File "C:\Python352\lib\threading.py", line 882 in _bootstrap

Current thread 0x00002f4c (most recent call first):
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 99 in show
  File "C:/buildbot/win-slave1/Cura2-windows/build/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 195 in event
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 195 in event
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 717 in run
  File "C:/buildbot/win-slave1/Cura2-windows/build/build/inst/bin/cura_app.py", line 70 in <module>
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cx_Freeze\initscripts\Console.py", line 26 in run
  File "C:\buildbot\win-slave1\Cura2-windows\build\build\inst\lib\python3.5\site-packages\cx_Freeze\initscripts\__startup__.py", line 14 in run

nallath, 2017-09-21:
I have no idea what the lulz-bot guys changed. I think you're better off asking them what could cause this. 
As far as I can see from the issue; you are running out of RAM.

fieldOfView, 2017-09-21:
Lulzbot are only building 32 bit versions of Cura for windows. A 32 bit application can only (easily) use 2 Gb of memory. Running out of RAM is one of the reasons Ultimaker is only building 64 bit versions.

Ghostkeeper, 2017-09-26:
There's nothing we can really do about it then though. Did you file this with Lulzbot?

ItzFitz2113, 2017-09-26:
This has been filed with Lulzbot. Thank you all for your help!

closed 2017-09-26 by ItzFitz2113