Ultimaker/Cura #2467: Bug Report requestd after crash

VinnyVG, 2017-09-22:
Version: 2.7.0-BETA
Platform: Darwin-16.7.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 490, in _onStackChanged
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 558, in _updateDisallowedAreas
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 481, in getUsedExtruderStacks
KeyError: '0'

-------------------------
Thread 0x0000700009933000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009430000 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008f2d000 (most recent call first):
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008a2a000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008527000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008024000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007b21000 (most recent call first):
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000761e000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000711b000 (most recent call first):
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

[... 48 lines cut ...]

Current thread 0x00007fff989e93c0 (most recent call first):
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 45 in exceptHook
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 536 in __globalContainerStackChanged
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315 in __performEmit
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 212 in emit
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/MachineSettingsAction/MachineSettingsAction.py", line 232 in forceUpdate
  File "/Applications/Cura 2.7.app/Contents/MacOS/plugins/plugins/MachineSettingsAction/MachineSettingsAction.py", line 182 in setMachineExtruderCount
  File "/Users/ultimaker/build/2.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 676 in run
  File "/Users/ultimaker/build/2.7/build/inst/bin/cura_app.py", line 73 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

fieldOfView, 2017-09-22:
Please update to the final Cura 2.7.0 release. This bug is already fixed.

VinnyVG, 2017-09-23:
Have done...thanks

closed 2017-09-23 by VinnyVG