Ultimaker/Cura #2468: Plugin system: Emitting a signal, after all plugins have been loaded.

thopiekar, 2017-09-22:
Hey all,

I know we are not able to resolve dependencies between plugins, but would it be possible to emit an signal after all plugins have been loaded or at least found?
Would be nice to have this, so I can make a check of all available plugins in my own plugin after loading (__init__ of them all) has been finished.

fieldOfView, 2017-09-22:
I use QtApplication.engineCreatedSignal for that. Strictly speaking it is a little later than after all plugins have loaded, but it works.

thopiekar, 2017-09-22:
Thanks @fieldOfView!

At least this is later, than what I'm looking for, which is totally fine.
The whole plugin management system needs more attention, in my point of view. Otherwise you can directly go back and use a filemanager again to handle your plugins.

fieldOfView, 2017-09-23:
Note that “engine” in this context is the QML engine, not the CuraEngine. The signal is emitted when the interface is about to be loaded.

fieldOfView, 2017-09-23:
The filemanager workaround will fail you if a plugin is disabled or fails to load.

thopiekar, 2017-09-23:
Come on @fieldOfView, what I expect over here are more features. At least hiding plugins, which are not compatible with an specific OS. Also what I need is something like a dependency resolving system. Eg. when installing a plugin, Cura should at least try to install other core plugins.
In case of the SolidWorks plugin I wrote, the same idea of converting a file (and this is a trivial one) can be used with other applications also. So why should I copy this commonized logic across different plugins I have? What I want is a general conversion plugin and make all the other plugins, which only include the API specific calls, depend on this one.

Thanks for the note. So as long as Cura does not change the logic of loading all plugins before loading the QML engine, then this signal should be always emitted afterwards.

Thanks again, @fieldOfView 😁

nallath, 2017-09-26:
Yeah, a plugin dependency system has been on the backlog for some time now. But if anything that i learned; Dependencies are *hard*.

thopiekar, 2017-09-26:
I can imagine.. guess it is easy to get a big mess..

I'm just curious how libsolv could help here, which is used by zypper, a package manager used for openSuSE 🤔

closed 2018-03-02 by ChrisTerBeke

fieldOfView, 2018-03-03:
CuraApplication.pluginsLoaded