Ultimaker/Cura #2469: Ironing Speed Problem

jonealex, 2017-09-22:
Wrong speed, as you can see on the video compressed here.

**Ironing enabled.**

[cura.zip](https://github.com/Ultimaker/Cura/files/1325347/cura.zip)

Ghostkeeper, 2017-10-01:
Huh, the travel speed is wrong apparently.

I don't see this though when I try to reproduce this. Here's a snippet of g-code taken from halfway through the ironing, where I had a sort of U-shape and it had to travel from one part to another:

    G1 X116.856 Y116.149 E2816.57716
    G10
    G0 F7200 X115.804 Y116.149
    G0 X107.196 Y116.049
    G0 X106.15 Y116.049
    G11
    G1 F800 X106.149 Y116.149 E2816.57751

The speed of the travel move shows as F7200, which is the normal travel speed of 120mm/s.

Maybe you can provide a project file that shows this problem?

ianpaschal, 2018-03-09:
Needs more info; never got it.

closed 2018-03-09 by ianpaschal