Ultimaker/Cura #2472: Cut objects

leobel96, 2017-09-23:
Please add the possibility to cut objects such as in slic3r

nallath, 2017-09-26:
I'll have a look as to how slic3r does it.

leobel96, 2017-09-26:
Perfect, thank you!!!

leobel96, 2017-09-29:
@nallath or maybe here https://github.com/kennethjiang/OctoPrint-Slicer/issues/133

leobel96, 2017-09-29:
And here https://github.com/kennethjiang/3tk/pull/5

nallath, 2017-10-02:
The one made by aleph objects seems to be a bit dead in the water.

nickthetait, 2017-10-02:
You mean non-fuctional or hasn't been updated in a while?

nallath, 2017-10-03:
A bit of both; It cuts the object but then it doesn't fill up the area that it cut, leaving you with an unprintable model. 
It's not the cutting that's hard, its the cutting and then ending up with two printable objects that is hard.

eyal0, 2017-11-11:
@nallath If it's useful for you, you can check my latest progress on https://github.com/kennethjiang/OctoPrint-Slicer/issues/135 .  I am able to cut the object and also fill the remaining hole pretty well!

closed 2023-07-13 by github-actions[bot]