Ultimaker/Cura #2473: Printing Speeds

jasonprovencher, 2017-09-23:
I am having issues with speeds in cura 2.7.0 on Mac. If you look at the screen shot, my speeds are pretty normal. As soon as it start printing, it appears that it goes faster over time AND not honoring the initial layer speed etc.

Am I missing a setting or is there a bug?

[image]

smartavionics, 2017-09-23:
Check that you have number of slow layers > 0.

jasonprovencher, 2017-09-23:
This setting is @ 2. Is there a setting that will accelerate the speed over time? All of my other settings are default.

smartavionics, 2017-09-24:
Not as far as I know. Could you please post a link to the gcode so that I can take a look at it?

closed 2018-03-02 by ChrisTerBeke