Ultimaker/Cura #2474: Cura 2.7 Fails to load any stl

stampedeboss, 2017-09-24:
Version: 2.7.0
Platform: Linux-4.4.0-21-generic-x86_64-with-debian-stretch-sid
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 713, in event
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 261, in event
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Event.py", line 189, in call
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 1319, in _readMeshFinished
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/Arrange.py", line 64, in create
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/ShapeArray.py", line 33, in fromPolygon
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/ShapeArray.py", line 70, in arrayFromPolygon
MemoryError

-------------------------
Thread 0x00007fc5d3fff700 (most recent call first):
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Backend/Backend.py", line 156 in _storeStderrToLogThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f09d5700 (most recent call first):
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Backend/Backend.py", line 148 in _storeOutputToLogThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f19d7700 (most recent call first):
  File "/tmp/.mount_rTBrgH/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f27fc700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f2ffd700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f37fe700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc5f3fff700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fc610dd9700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 293 in wait
[... 80 lines cut ...]

Thread 0x00007fc63799f700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007fc65a5af700 (most recent call first):
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/CrashHandler.py", line 100 in show
  File "/build/2.7/build/inst/bin/cura", line 45 in exceptHook
  File "/build/2.7/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 682 in run
  File "/build/2.7/build/inst/bin/cura", line 73 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/2.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

nallath, 2017-09-26:
You are running out of RAM memory. There is little we can do about this i'm afraid :(

closed 2017-09-27 by ChrisTerBeke

stampedeboss, 2017-10-05:
According to the OS (linux) I have 13G available.  Additionally, it works with Cura 2.6.2.  Running on a Mint KDE version, system has 16GB memory and this is the only application running.

Ghostkeeper, 2017-10-15:
So this bug is in the Arranger code, but nothing really changed between 2.6 and 2.7 with that bit :confused: