Ultimaker/Cura #2476: [REQUEST] First Layer Retraction Settings

dmezh, 2017-09-24:
Howdy - I think Cura should have separate retraction settings for the first layer - specifically, we should have the option to turn off combing for the first layer only.

This is very useful because perimeters, especially small or circular ones, get pulled away from the bed when the nozzle travels on the first layer. It's not an adhesion issue. In addition to the remnants of the perimeter that has been pulled away, the nozzle can also leak or scratch existing surfaces. As a result, the next perimeter after the travel will be underextruded because some plastic has leaked out.

[image]
[image]

Adding the option [Always Retract on First Layer Travel] (and, if you're feeling ambitious, a [Maximum Combing Distance on First Layer] to prevent retracts on tiny travels) would be immensely helpful.

Ghostkeeper, 2017-10-01:
But combing should _prevent_ hitting the walls of the first layer, right? Why would turning it off be better?

dmezh, 2017-10-01:
The effect is pretty much the opposite. Especially on circular walls, the nozzle will move away without a retraction as soon as it is done with the loop. This has two main effects. The first, and most important, is that the wall can be pulled away from the bed. 

This is especially bad and easy to reproduce because Cura behaves differently from other slicers in that it doesn't print walls in local groups, but rather doing each wall number at a time everywhere in the object. It's not an adhesion issue, but more so that these situations are torture tests for first layer behavior.

The pulled-away walls can mess up other parts of the first layer. Additionally, oozing from the nozzle can mess up the first layer, particularly so on tiny details.

These issues are what are causing the problems in the images. Always forcing a retraction (perhaps with a minimum travel setting as well) on first layer travel will greatly lessen or entirely solve this problem. Additionally, it would improve first layer quality, even on successful first layers with simple geometry and no curves.

This behavior is less of an issue on non-first layers, because plastic-to-plastic adhesion is strong and instant, and travels can be done faster.

Ghostkeeper, 2017-10-11:
I'll file that as a feature request then, to always retract on the first layer. I can see that working better.

cheesecakegangster, 2017-10-13:
would also like this feature for the last layer(s)

Ghostkeeper, 2018-01-15:
Our project manager just removed this from our planning as it was more than 12 weeks old. Looks like we won't be implementing this any time soon.

sisu-jason, 2018-01-24:
We are having the same issue the OP described with a 6in+ diameter part with small holes around the perimeter and a larger hole in the middle. Because there is no retraction with travel moves on the first layer, a bit of material strings from the nozzle during travel moves. This will eventually gum up around the nozzle, then adhere to the outside wall of one of the small holes and drag that around too. From there the print is failed. Retraction during travel moves would most likely help. Not sure what else to do except maybe print the whole thing on a raft (which seems like a waste of extra time/material for a part that should print fine on the plate). 

This is with an Ultimaker 3 extended, Cura 3.1, white Ultimaker PLA, started with all the standard settings.  We cleaned the build plate, then tried using the gluestick that comes with the printer (for the first time ever). Then We tried lowering the nozzle temperature a bit. Nothing really worked...

Sorry, no pictures on the build plate, but you can see the string/dragged walls:

[image]

[image]

jackha, 2018-01-25:
@sisu-jason I'm sorry to hear you have these problems. Can you set Combing Mode to Off and see what happens then? In addition you can try Z Hop When Retracted. The downside is that you can get stringing in the holes and the Z Hop makes the print slower.

george-vieira, 2018-06-19:
+1
This is a critical setting. Turning it off completely makes print jobs take a hell lot longer all because of the first layer. Now all layers have to retract which slowed the print time down.
The print quality and stability key to 3D printing, how is this dropped off the planning list?

ZhuJo, 2018-07-12:
+1 
This really sucks. I'm laboring for more than a week over this issue until I found this. Why Cura doesn't retract on first layer? I would somewhat (althought hardly) swallow the crosslines on the skin, but I can't get over the fact that it sometimes drags whole first layer wall with it.

This really should get highest priority. I can't use a Slic3r as it overextrudes like crazy and I don't want to turn each setting here and there. For me, a slicer should be doing only one thing only - getting the best quality prints within a reasonable time. I don't care about that setting there and here. Setting the material, infill and it should be good to go.

FredGenius, 2018-08-06:
I agree completely with the OP. I also print on glass, and often need the first layer to be perfect as it will be glassy smooth and the lines produced when the nozzle leaks filament during travel moves are highly visible. I've even resorted to editing the gcode manually, very time consuming. Please implement this feature, I like Cura but this is a really annoying problem.

gladfelter, 2018-08-12:
+1
I'm seeing the same thing on un-heated glass. I'm going to try the "Not in Skin" option for the "Combing Mode" configuration option and see if that helps at all.

FredGenius, 2018-08-12:
I think I tried that, Seems Cura thinks 'because there is no skin already, it's ok not to retract'. But please go ahead and try, I have a bad memory and I've been known to be wrong.

smartavionics, 2018-08-12:
Folks, I currently have a PR submitted that fixes the "no skin" combing so that it doesn't comb over skin or the gaps between parts on the first layer. I believe that it may fix the problems described above. See https://github.com/Ultimaker/CuraEngine/pull/781 if your interested in the details.

smartavionics, 2018-08-12:
> Many thanks, any idea when we may see this feature included?

Sorry, haven't a clue. The PRs can languish for months before they get around to looking at them and deciding if they want to use them or not. I'm afraid the Cura devs have different priorities to the rest of us.

jackha, 2018-08-13:
There was an JIRA issue created for the pull request, but unfortunately I forgot to put it in the sprint. Thanks to activity here I looked it up and it's back on our radar again :-)

FredGenius, 2018-08-13:
Thanks for the update!

DIYglenn, 2018-08-17:
I would love a fix for this. 
I’ve had a a lot of problems with my prints and haven’t understood why. Most of my prints involves holes/squares/honeycomb so this has made my printer useless - no good prints. 
Will test the “No skin” setting for Combing (disable for visible top and bottom layer).

smartavionics, 2018-08-17:
If you have access to a Linux system you can test it today using my releases which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

DIYglenn, 2018-09-02:
@smartavionics Late reply here, but Yes - I do have multiple Linux systems, and will definitely test this if it isn't pulled in the near future. 
Having combing completely off seems to slightly increase the chance of stringing, but definitely solves my problems on the first layer. NEVER had a part come off after this. 

What exactly is the difference of the combing mode "Not in skin"? Does that affect the first layer at all?

smartavionics, 2018-09-02:
> What exactly is the difference of the combing mode "Not in skin"? Does that affect the first layer at all?

The original not in skin combing mode was intended to avoid dragging the nozzle across skin areas by only combing within the part's infill regions rather than within the part's outline. Unfortunately it was badly flawed in that it only took into account the outline of the part's infill areas and took no notice of the part's skin and walls so would happily cross those leaving nozzle marks/dribble. 

I have reimplemented the no skin combing mode so that it only combs within infill and on the part's walls. So now it will not comb over skin and it will do retracts when moving from part to part. The original no skin combing behaviour will still be accessible under a different name.

FredGenius, 2018-09-02:
Thanks smartavionics, that's great news!

FredGenius, 2018-09-03:
Thanks nallath, I look forward to that.

bblack7489, 2018-09-29:
thanks @nallath 
This is great news, as I have been struggling with this on a part as well.  I see the behavior that you described as "flawed," in that I have a part that's basically like a cup.  When the "not on skin" mode is on, it treats the inside surface of the cup just like it was infill.  The end result has some artifacts on the inner surface.
I'm looking forward to the update as well.

closed 2018-11-23 by Ghostkeeper