Ultimaker/Cura #2478: exception in stl.py "could not convert string to float: b'-1.#ind'"

mkobzar, 2017-09-25:
cura cannot open some stl files, sample is attached

[Scans.zip](https://github.com/Ultimaker/Cura/files/1328105/Scans.zip)

2017-09-24 21:46:47,601 - ERROR - UM.Logger.logException [76]: Exception: Exception occurred while loading file C:/Users/mike/Documents/Scans/model1000faces.stl
2017-09-24 21:46:47,603 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-09-24 21:46:47,605 - ERROR - UM.Logger.logException [80]:   File 
"D:\2.7\build\inst\lib\python3.5\site-packages\UM\FileHandler\ReadFileJob.py", line 61, in run
2017-09-24 21:46:47,607 - ERROR - UM.Logger.logException [80]:   File "D:\2.7\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshFileHandler.py", line 28, in readerRead
2017-09-24 21:46:47,609 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\FileHandlers\STLReader\STLReader.py", line 56, in read
2017-09-24 21:46:47,611 - ERROR - UM.Logger.logException [80]:     self.load_file(file_name, mesh_builder, _use_numpystl = use_numpystl)
2017-09-24 21:46:47,613 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\FileHandlers\STLReader\STLReader.py", line 35, in load_file
2017-09-24 21:46:47,615 - ERROR - UM.Logger.logException [80]:     self._loadWithNumpySTL(file_name, mesh_builder)
2017-09-24 21:46:47,617 - ERROR - UM.Logger.logException [80]:   File "C:\Program Files\Cura 2.7\plugins\FileHandlers\STLReader\STLReader.py", line 87, in _loadWithNumpySTL
2017-09-24 21:46:47,619 - ERROR - UM.Logger.logException [80]:     loaded_data = stl.mesh.Mesh.from_file(file_name)
2017-09-24 21:46:47,621 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 294, in from_file
2017-09-24 21:46:47,623 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 55, in load
2017-09-24 21:46:47,625 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 199, in _load_ascii
2017-09-24 21:46:47,627 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 181, in _ascii_reader
2017-09-24 21:46:47,629 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 152, in get
2017-09-24 21:46:47,631 - ERROR - UM.Logger.logException [80]:   File "stl\stl.py", line 152, in <listcomp>
2017-09-24 21:46:47,633 - ERROR - UM.Logger.logException [80]: ValueError: could not convert string to float: b'-1.#ind'

mkobzar, 2017-09-25:
I just realized that my stl which (exported by Skanect) contains 264 instances of "-1.#IND" strings which suppose to be floats.
example - "facet normal -1.#IND -1.#IND -1.#IND"
... not sure you could ignore / workaround with them

nallath, 2017-09-26:
It's actually an invalid STL file.

closed 2017-10-18 by ChrisTerBeke