Ultimaker/Cura #2485: When there is only 1 quality, Recommended mode always shows "slower"

fieldOfView, 2017-09-26:
With the new quality selection slider, if there is only a single available quality, the slider is always showing a "Slower" print speed. This will not look great when the user picks a 0.8mm nozzle for their Ultimaker 2+ (you would think that a 0.8mm slice at 0.2mm layer height is not "slow").

From a UX perspective, it would be better if the disabled "nub" that is visible when there is only a single layerheight is in the center (neither slower nor faster).

nallath, 2017-09-27:
Put it in our current sprint; CURA-4383

Appesteijn, 2017-09-27:
Just discussed this and it might be to difficult with the Qt slider (especially before the 3.0 launch)

fieldOfView, 2017-09-27:
FTFY

ChrisTerBeke, 2017-09-27:
Noice, reviewing right now

ChrisTerBeke, 2017-09-27:
There's an issue where if you change the material of the extruder back to something that 'unlocks' more values on the slider, the value doesn't update and the slider handle is in the middle between 2 tick marks (for example: PVA + PVA -> ABS + PVA).

ChrisTerBeke, 2017-09-27:
Closing now that issue has been picked up internally and has a PR.

commit 3be8f456 (2017-09-28): Move the quality indicator to the center when there is only one quality available

Fixes #2485
CURA-4383
PR #2500 (2017-09-28): Move the quality indicator to the center when there is only one quality
This PR fixes a UX issue where if only a single quality is available, it always shows a "Slower". If a single quality is available, the quality indicator (handle) is now moved to the center of the slider.

Fixes #2485
CURA-4383
closed 2017-09-27 by ChrisTerBeke