Ultimaker/Cura #2488: temperature settings "stuck" after using Per Model Settings

yourhead, 2017-09-26:
Days ago I created a project test speeds with some new filament. The project had several models. Each model had some "Per Model Settings".

Since this time all gcode produced outputs to the same temperature setting I used for that run:  245°.

I've tried changing everything including temperatures, hiding/showing more settings, re-creating the multi-model project -- but nothing seems to have any affect on the produced gcode.

To be 100% clear: the "Printing Temperature" setting is currently set to 195. But the output gcode continues to be 245.

I'd be happy to provide logs, gcode files, etc if you let me know what you need to investigate further.

nallath, 2017-09-27:
Could you share the project file with us? That makes it a lot easier to follow what you did exactly.

yourhead, 2017-09-27:
Sure.
https://www.dropbox.com/s/ztufmngvuh45a3e/Project.zip?dl=0

To be clear:  This behavior exists on ALL projects now. It is not associated with any single project.

So, I've just created a project from scratch with an STL file I had around. I've included the project file (.3mf), the gcode that cura writes out, as well as the profile that i used (curaprofile).

The profile in question was not overridden in the UI. It displays as having a temperature of 195º. If there is some other hidden temperature that supersedes this -- it would probably (especially for temperature) be a good idea to display what the user is going to get **somewhere** pretty obvious.

in other words:  if the answer to this is that i have some hidden property. then this bug report is "that property should never be hidden"

In my case the gcode shows 245
[image]

nallath, 2017-09-28:
I've just tried this with master (on a Custom FDM Printer). 
It seems to work just fine; 

[image]
[image]

Could you share me the project file and not just the .3mf file?

closed 2017-09-28 by yourhead