Ultimaker/Cura #2489: Right-Click "stutters" when dragging.

yourhead, 2017-09-26:
At least on macOS the Right-Click is bound to both the rotate the camera view and to showing the contextual menu.

It seems like the desired effect is that if the mouse is held in place without moving the context menu pops up -- and if the mouse is moving you get the rotate behavior.

1.  If I right click and drag -- the rotate behavior doesn't start immediately.  There's a substantial lag that makes it feel like i've clicked the wrong thing. Give that every 3D modeling app maps the keys a bit differently -- the lag is enough to make me pause and wonder if my fingers have done the right thing.

2. With some HID devices it can be quite difficult to get the context menu to pop.  When I'm using my trackpad -- it has a very sensitive input. It's very difficult to get the cursor to be still enough for the context behavior to occur.

My recommendation would be to:

1. remove the stutter from the click-drag to rotate if at all possible.
2. move the context menu to another mouse button -- or make it configurable.

nallath, 2017-09-27:
We use a variable called "drag_distance", which is set at 0.05. So movements above that are seen as either a move/rotate if so required. I'm not noticing any issues with it feeling laggy. 
If you want to have a go at changing this; Change the self._drag_distance to something lower in plugins/CameraTool/CameraTool.py. 

I don't have a mac myself, but I was under the impression that ctrl+click should also register as a right mouse click. So that would at least for the moment lessen the impact.

As for re-mapping certain events, this is something that I really want; A number of plugins should also be able to hook themselves up to shortcuts. But as these things go, there are always to much features and too little time.

ChrisTerBeke, 2017-09-27:
I can confirm this behaviour on Mac, however the delay is hardly visible for me.
I would say the context menu should remain on the right mouse button, as that's the case with any piece of software out there.

fieldOfView, 2017-09-27:
> I would say the context menu should remain on the right mouse button, as that's the case with any piece of software out there.

Agreed. I have always wondered why we don't use the left mousebutton to rotate the camera though

ChrisTerBeke, 2017-09-27:
Indeed most 3D software uses click and hold of the left mouse button for rotating and right button for moving. Maybe this part of the behaviour could be user configurable (but that should be a separate ticket).

Ghostkeeper, 2017-10-01:
In my experience, all 3D software uses their mouse buttons differently 😛

If we were to allow customising this, how would we expose this in the GUI? Via a drop-down that allows the user to choose between "Like SolidWorks", "Like Blender", "Like OpenSCAD", etc?

ianpaschal, 2018-03-05:
> If we were to allow customising this, how would we expose this in the GUI? Via a drop-down that allows the user to choose between "Like SolidWorks", "Like Blender", "Like OpenSCAD", etc?

That's what FreeCAD does.

comps, 2018-06-22:
I noticed this as well, switching from an older version of Cura to 3.3.0. The right-click "deadzone" is very significant, making it hard to do precise rotations to inspect a tiny feature on a model and just makes the UI feel laggy and unresponsive. I haven't encountered this in any other 3D software.

closed 2018-08-10 by diegopradogesto