Ultimaker/Cura #2492: Editable text scrambled when using Retina display

yourhead, 2017-09-26:
If I use Cura 2.7 on a retina display (HiDPI) the display is very scrambled.

If I switch the display resolution to a non-HiDPI resolution then the problem (mostly) goes away.

Here's a screenshot of what I see on a retina display (in dark mode -- it's similar in light mode too)
[image]

nallath, 2017-09-27:
@Appesteijn Could you check if you also have the issue?

Appesteijn, 2017-09-27:
It almost looks like the selected font is wrong. This is how it looks on my Retina display:
[image]

Did you change anything with your system fonts? This is my setup:
[image]

ChrisTerBeke, 2017-09-27:
I am not seeing this behaviour on my Retina MacBook Pro, neither in 2.7 or 3.0/master. Maybe a used font is missing on @yourhead's machine?

ChrisTerBeke, 2017-09-27:
Related to https://github.com/Ultimaker/Cura/issues/2462?

yourhead, 2017-09-27:
This is our macOS 10.12 test machine. It's configured with the default setup.

Other Qt based software (Fusion 360, Simplify3D) behaves as expected on the same hardware and setup.

Here is a screenshot of the General Prefs as req'ed. This is a test-machine for our own software, it should be the default for macOS 10.12.6 (for this hardware, i.e. no handoff, no trackpad)

[image]

Appesteijn, 2017-09-27:
Could you try without 'Use LCD font smoothing..'?

yourhead, 2017-09-27:
same behavior.
[image]

i don't think you guys are using the macOS font rendering. changing this setting immediately affects all fonts in every app…    except Cura.
😝

One interesting big:  it only affects some fonts in the display, not all fonts (I outlined in red). I'd look for what is similar or different about those. Maybe a different font? Or maybe it's something more insidious like the bounding boxes of those text areas are landing on non-integer pixel bounds when dimensions are divided by 2? Anyway, that's where I'd look. Good luck!  😃

yourhead, 2017-09-27:
Some other interesting tidbits:

- each character that is rendered incorrectly is **consistent**. e.g.: in the above example the lowercase "h" character is always rendered as just the tip of its ascender. Then the rest is rendered correctly.
This leads me to believe the bug would be in the font-handling -- converting character bitmaps/shapes into rasters.

- i can ask macOS to zoom the entire display using the assistive tools. this zooms things on the video card -- it's just an affine transform of the entire display's bitmap. this sometimes exposes problems in low-level video card/opengl stuff.  but in this bug the display is consistent -- the bad characters are zoomed along with the rest of the display.

- other similar items display properly:
the "mm" in each edit field is displayed correctly. as are the PopUp control labels "ZigZag" -- this behavior is consistent:  all popups display perfect no matter their content. All the unit-labels display correctly.

- each launch is different. sometimes i get lucky and the text is quite usable. other times it's scrambled beyond belief. Here is another launch. Note that the numbers are A-OK this time -- but I've lost lowercase "p" in the blue help text.

[image]

Appesteijn, 2017-09-28:
Do you also have this when you use the default theme?

ChrisTerBeke, 2017-09-28:
The UI is buid with QT. According to their documentation, the MacOS implementation uses Cocoa for font rendering (http://doc.qt.io/qt-5/qfont.html), should be native right?

fieldOfView, 2017-09-28:
QML Label uses native font rendering, QML Text uses non-native text rendering.
We used to use only Label elements, but since a while ago Text and Label are used intermingled :-/
https://github.com/Ultimaker/Cura/issues/2068#issuecomment-324917547

Recently I made a couple of PRs that fixes the ugly font rendering in all "system-themed" dialogs, by reverting back to using Label and making sure the texts are rendered at whole-pixel intervals. I think the best fix would be to do the same for all "Cura-themed" texts too (ie: revert back to using Label over Text, and making sure the elements are placed on whole pixel intervals).

ChrisTerBeke, 2017-09-28:
If time allows we'll look into the Text vs Label situation and try to find every occurrence of it. Another solution might be implementing a UM.Text (or UM.Label) component so we can have a central place to make any future changed needed to font rendering.

fieldOfView, 2017-10-02:
/sa https://ultimaker.com/en/community/51334-cura-fonts-garbled-on-macos-high-sierra

Appesteijn, 2017-10-03:
I tried to make fix for this: https://github.com/Ultimaker/Uranium/pull/287

ianpaschal, 2018-02-27:
This will be fixed in the next release of Cura (3.3).

closed 2018-02-27 by ianpaschal