Ultimaker/Cura #2493: Dragging the Cura window from a Retina display to a non-Retina display has unexpected effects

yourhead, 2017-09-26:
Using macOS with two displays: one retina and one standard (a very common setup for any user with a recent model macBook and an external display).

If you open Cura on the retina display then move the window to the standard display (e.g. you open it on your MacBook internal display and then drag it to your larger external monitor)…

The camera view seems to zoom in. Pushing the origin of the workspace to the upper-right. However it is not merely zoomed. Cura still thinks the origin is way up there off the screen somewhere. So Cura is unusable.

There are workarounds to get cura over to the other monitor by launching it with different display setups of the MacBook -- but it's quite the pain. Would be great if it just behaved like other apps.

Here are a couple screenshots.  The first is on a retina display (note the scrambled fonts -- that's another bug also filed):

[image]

and here's the same window after dragging it to the external display:

[image]

Note the model, which is centered in the workspace is zoomed off to the upper left.

nallath, 2017-09-27:
@Appesteijn; Could you check if you could reproduce this?

Appesteijn, 2017-09-27:
We are currently implementing a fix for this, it is expected to be put live at 17 October (3.0.0). But I think we still only calculate the screen pixel density at startup, so dragging Cura to another screen might use a wrong DPI setting.

fieldOfView, 2017-09-27:
On OSX it the DPI setting is always the same between all displays. OSX renders everything at the DPI that is the highest between the connected displays, and then scales the window on the other DPI screens transparently. However, it is very likely that the opengl viewport does not "like" this.

ChrisTerBeke, 2017-09-27:
As this only affects a very specific edge case where you drag between different DPI monitors, and only on Mac, I think it's too small to fix. I'm not even sure if we can detect whether a window has changed to another monitor and then do something with that info.

yourhead, 2017-09-27:
NSWindowDidChangeBackingPropertiesNotification when properties like the screen depth of a window change.
https://developer.apple.com/documentation/appkit/nswindowdidchangebackingpropertiesnotification?language=objc

Both NSWindow and NSScreen have a backingScaleFactor property that provides the specific depth information about any window or screen.
https://developer.apple.com/library/content/documentation/GraphicsAnimation/Conceptual/HighResolutionOSX/APIs/APIs.html

ChrisTerBeke, 2017-09-27:
Please note that we're using Python and currently have not implemented any way to use Cocoa native elements. I've seen several solutions to get appkit libraries connected to python but that's a bigger change in the code base that needs internal discussion first. Will leave this ticket open for future reference.

awhiemstra, 2017-09-28:
The relevant Qt API you are looking for is http://doc.qt.io/qt-5/qwindow.html#screenChanged . Still, I think it would be better to try and get the Qt 5.6 high-DPI support working, since that should support things like this natively.

ianpaschal, 2018-05-08:
This has seen no activity for 7+ months and is not really a big problem. There's no glitch or crash or freeze or anything; just gotta zoom out a bit. I don't think we'll bother fixing this.

closed 2018-05-08 by ianpaschal