Ultimaker/Cura #2495: Suggestion: Expand/Collapse button for print setup menus

Thisismydigitalself, 2017-09-26:
Right hand side settings menus should have an Expand/Collapse All button.
If there is one and I've missed it - please come forward. It would also be nice to have favorites for quick Find & Edit of selected features.

fieldOfView, 2017-09-27:
>  It would also be nice to have favorites for quick Find & Edit of selected features.

Please clarify this

Thisismydigitalself, 2017-09-27:
To be able to select out of many parameters one uses most.
I have marked few input settings in RED to illustrate a nice-to-have option to select those to be included in say - favorites-1 group. basically narrowing down the long list and group most used parameters in a logical way than saving complete profile. it's more like filtering a profile.
[image]

Ghostkeeper, 2017-10-01:
We'd like to implement visibility presets some day. Custom presets is a step beyond that then.

Thisismydigitalself, 2018-02-14:
Cura 3.2, still no Expand/Collapse for settings menus.
If each category in the settings would have opened up in a different tab like windows file properties window then there was no need for Expand/Collapse button.

Ghostkeeper, 2018-03-07:
The visibility presets are in for 3.3 though. It's not possible yet to save custom presets but there has been some progress.
[image]

Thisismydigitalself, 2018-03-07:
Progress indeed. Please add this very basic, simple and needed expand/collapse for all print setup settings of GUI's right pane so when you have few categories open, you can click and collapse all for faster navigation. should be very simple to implement. thank you. 

[image]

ianpaschal, 2018-03-09:
> should be very simple to implement.

Ugh. Please don't say this. Very little is ever simple to implement. If it's simple, feel free to make a pull request.

Personally, I think the setting visibility profiles solves this issue for now.

Thisismydigitalself, 2018-03-09:
Oops, sorry for stepping on so many toes. i mean no harm.
I don't understand how profile visibility is going to work as i never used it before (looking forward) but whenever i start a new print/project i go over many settings under many categories and program my slicer for best result. since the mechanism allows us to open many categories simultaneously it was nice to have a button to collapse all which will make it easier to find and reopen other categories to edit.
I am not a programmer (should be obvious by now) but a devoted Cura user so i only speak from experience.

ianpaschal, 2018-03-09:
> I am not a programmer (should be obvious by now) ... i mean no harm.

I know and I know. But that's why I say what I did. I can imagine it looks obvious to you but so you know: it usually isn't. Even as programmers we often underestimate the difficulty of problems because code is akin to a biological ecosystem and changing things in one place tends to throw the system off balance and cause issues other places. The better code we write, the less chance there is of that, but if we ever figure out how to write 100% bug free first-time-right code then we should be teaching computer science at Stanford, not working at Ultimaker. :P

So yeah. It's not that easy.

Luckily, this new feature which I take it from  your comment hasn't been shipped yet (I loose track a bit from the time delay between when we work on something and when it gets shipped), switch between 3 setting visibility presets (basic, advanced, and expert) as well as save custom selections of options. The presets are pretty basic config files (just lists of setting names) so in theory we could do a lot more with it but already it was a trick to implement and it's extremely tedious to update by hand. Simple concept, PITA to keep up-to-date. It will improve over time, though.

Thisismydigitalself, 2018-03-10:
[image]

jackha, 2018-08-16:
Now it's shipped for sure ^^.

closed 2018-08-16 by jackha

Thisismydigitalself, 2018-12-20:
It seems that Cura 4.0B not only missing this basic fearure but is a step backwards from 3.6 by having even shorter list of settings and no collapse / expend all settings. 

Sorry mates, i just don't get this.