Ultimaker/Cura #2497: Cura 2.7: Missing option Print Sequence

tthurow, 2017-09-27:
There is no option to set the print sequence by more as on model.

[image]

[image]

nallath, 2017-09-27:
This has never been possible on the UM3 (or any dual extrusion machine)

tthurow, 2017-09-28:
Then the error message is misleading. I also do not understand this limitation, especially in the industrial environment, this is an extremely important function to be able to print parts sequentially onto the printing plate, unattended over hours. In my view there is no technical reason against the function, only the restricted areas are larger.

fieldOfView, 2017-09-28:
You can in fact go to Preferences -> Printers -> Machine Settings, and set the number of extruders to 1. Then you can print One at a Time, though only with the first extruder.

Ghostkeeper, 2017-10-03:
The limitation is a technical one that could in theory be resolved in Cura but we haven't yet. There's two things preventing us from allowing One-at-a-Time printing with multi-extrusion right now:
* When moving to a wipe brush or switching bay, you might hit previously printed objects.
* You need multiple prime towers.

The error message is a bit unclear because it's automatically generated, but it's the number of extruders that you have that prevents this setting from being visible, so the message by itself is correct.

ianpaschal, 2018-05-08:
One at a time mode is now working thanks to disabled extruders.

closed 2018-05-08 by ianpaschal

Ghostkeeper, 2018-06-07:
To be clear, you can now disable extruders so that there is only one extruder left. Then you'll be able to select one-at-a-time mode again.

However dual extrusion one-at-a-time printing still doesn't work. We have an issue open for 3+-extruder prime towers.