Ultimaker/Cura #2501: 2.7 Fails to load STL file from Blender 2.87c

tclancey, 2017-09-27:
Complex or really simple models fail to load.  That's all it says, Failed to load file.  

So I got my new Ultimaker setup at 11am this morning and I've been trying to get a test file out from Blender since.  It's now 16:33.

I think I'm right in saying Blender hasn't changed the STL export for some time, can someone please shed some light on the problem for me?

Thanks.

smartavionics, 2017-09-27:
Hello, could you please post a link to a failing STL, then it could be investigated.

nallath, 2017-09-28:
Could you also share the logs?

Ghostkeeper, 2017-10-04:
Can't reproduce with the Blender monkey: 
[image]

Works fine here.

DDDirk, 2017-10-04:
I just sliced (and printed) a model sliced in Blender 2.78a, which worked just fine. So I wanted to download version 2.87c. Which doesn't appear to exist. Did you get that version number correct? Well, there is 2.8, but that is experimental.

Ghostkeeper, 2017-10-15:
Here is 2.78c: https://www.blender.org/features/releases/2-78/

DDDirk, 2017-10-16:
Ok, so I downloaded that (more specifically blender-2.78c-linux-glibc219-x86_64) and again, in Cura 2.7 it opened and sliced just fine.

DDDirk, 2017-10-17:
Have you tried opening it in another program (a 3D viewer)?
Also, in Blender you can install the 3D Print Toolbox. To install: Crt-Alt-U > Add-ons, in the search box enter 'print', select the tool and 'save user settings'. It then appears in a tab at the left. Click that and 'check all' and then if necessary 'make manifold'. That might solve it.

closed 2017-11-14 by ChrisTerBeke