Ultimaker/Cura #2502: Free floating print points

Thisismydigitalself, 2017-09-27:
This happens to me all the time: 
Slicing an object leaves free floating islands of points which by then the nozzle oozes small amount of filament and jumps from on to the other in an unpredictably manner. why the algorithm does not move the nozzle in strait lines to overcome this somewhat silly behavior? this happens when infill is @ 0 and i fill the part by choosing enough wall count to overcome gaps. it seems better in 100% infill but than the weaving pattern is not like what i wish for. Any workaround?
[image]

smartavionics, 2017-09-27:
Hi. The next version of Cura will incorporate an experimental optimizer that reduces the amount of travel when printing prints like your example. When the software is released, please try it out and let me know if it solves your problem.

Thisismydigitalself, 2017-09-28:
Will do.
Are we talking 2.7.1 or 2.8?

fieldOfView, 2017-09-28:
3.0

Thisismydigitalself, 2017-10-10:
I have installed 3.0.
Please direct me to which experimental feature i should use to solve this line issue.

Thank you.

Thisismydigitalself, 2017-10-10:
Just tried "Remove all holes" which seems to do the trick.

smartavionics, 2017-10-10:
The feature is called "Optimize Wall Printing Order". Please be aware that there's currently a bug that can cause crashes if infill before walls is enabled as well as the optimization. Sorry about that!

Thisismydigitalself, 2017-10-10:
Sorry to say this does not fix my issue. 
when i enable "Remove all holes" it fixes this issues but at the same time removes all holes from model. not good.

smartavionics, 2017-10-10:
Could you please make the model available (post a link to the STL or upload to github as a text file). I could then get a better understanding as to what is happening.

Thisismydigitalself, 2017-10-10:
[AEROSOL v6.zip](https://github.com/Ultimaker/Cura/files/1372370/AEROSOL.v6.zip)

smartavionics, 2017-10-10:
Thanks for the model. If I slice that with 6 walls the print time estimate is 37 mins without optimization and 34 minutes with. That's OK, isn't it?

Thisismydigitalself, 2017-10-10:
The problem is not time. the problem is cura's slicing algorithm. why we get floating islands like in the first picture i attached at the beginning of this thread?

smartavionics, 2017-10-10:
Have you tried using the Outer Before Inner Walls setting? That may be your best option. Perhaps without optimization. Otherwise, I don't know what to suggest. It has to fill those areas somehow.

Thisismydigitalself, 2017-10-10:
> Have  you tried using the Outer Before Inner Walls setting?

> 

yes, no change.
I believe this issue should be reopened. we need to be able to print such models without infill but with enough walls to fill the entire model and have cura slice those without any islands.  when there are islands it take more time to print, the head jumps from one island to the other and all over the place and print is not smooth.

Thisismydigitalself, 2017-10-10:
ChrisTerBeke  closed it. what's the mechanism to reopen closed issues?

smartavionics, 2017-10-10:
How can that shape be completely filled with walls without the islands appearing?

@ChrisTerBeke  should see our messages.

ChrisTerBeke, 2017-10-10:
Tagging me like in the message above is enough :) (Policy is: when a repo editor closes an issue, only other editors can re-open it).

smartavionics, 2017-10-10:
Hi Chris, I'm willing to help with this issue but struggling to understand exactly what the OP is requesting here.

ChrisTerBeke, 2017-10-10:
I think it's something like "don't print walls when line is shorter than x", which becomes very obvious when the whole print is basically walls.

Thisismydigitalself, 2017-10-11:
Hi. thanks for reopening this issues.
Yes, "don't print walls when lines are shorter than x" sounds like a good way around this.
I was thinking more in the direction of continues spiral to the very center core to the end of the line/wall but I trust you guys to figure this out the best way you know how.  don't print shorter than X might end up leaving small holes as oppose to how it is been printed these days.

You can try this yourself: 
When printing only walls without infill it takes less time and better looking print with much less nozzle's awkward dance all over the place.

Thank you.

Ghostkeeper, 2017-10-19:
I think that the Optimize Wall Printing Order removes a big part of the problem, but you can't actually see the change very well in Cura's layer view.

Removing short walls may delete details though. If you print a spike it will get "stumped" because the top of the spike has too short walls. Inverse, if you have a corner of air pointing downwards it will get a wide hole with a flat floor.

DDDirk, 2017-10-20:
The spike problem might be solved by not applying it to outer walls.

Thisismydigitalself, 2017-10-21:
15 posts and I don't yet know how to solve this problem.
I was told 3.0 will solve this issue but to this point in time I have no idea how to overcome the short lines / islands).

smartavionics, 2017-10-21:
The problem I have with this is that I don't understand what you are asking for. In the model shown in the picture at the top of this thread how can you fill that shape with walls without having some small line segments? Please provide a picture or sketch showing how you think those walls should be printed.

Thisismydigitalself, 2017-10-22:
I was thinking and already stated this in this thread, to just continue in a closing spiral all the way to the very core and not break the line and deposit small islands of blobs.  I bet that if you would print with only walls you will see what I'm talking about. it just makes no sense as it is now.

[image]

smartavionics, 2017-10-22:
I still don't see how the shape above can be filled as a spiral because of the various holes and the fact that some regions get narrow and other regions get wider, etc. You are always going to be left with little areas that need to be filled separately like the areas that you highlight above. And those areas are currently filled with a number of concentric lines that will, in effect, be printed as a spiral with, say, one travel move to the little area to be filled, the extrusion moves that fill the area and then a travel move away from the area.

Thisismydigitalself, 2017-10-22:
are you saying it is not possible to continue the line all the way in and you must break it and print a floating blob in space?

are you saying this (in red) is not possible?
[image]

smartavionics, 2017-10-22:
At the moment it does this:

[image]

It travels to the top right corner, extrudes in an arc to the bottom left, bottom and then back to the top right, it then moves to the middle (no retract used so the plastic will keep oozing during the move) and then it extrudes the small middle line and after that it does a retraction and moves away from the region. Not sure what that weird little kinky travel move is under the middle line, though. Anway, the point I am making is that the end result is not going to be much different if that little region was printed as a real spiral as the only move between extrusions is very short and the extruder will keep oozing anyway.

So it could be possible to print that region as a spiral but would it really produce a noticeably different end result? I'm not sure that it would. With a spiral, when it gets to the middle it still has to retract and move away and that's going to leave an artifact. You can't avoid them completely.

Thisismydigitalself, 2017-10-22:
> So it could be possible to print that region as a spiral but would it really produce a noticeably different end result? I'm not sure that it would. 

> 

Please print it and you'll see what I'm talking about. every layer has a bunch of these. for each the nozzle runs to and fro and deposit a blob. add to that the nozzle retraction and you'll see how a simple print turns out to be messy for no real reason. i use this method cause printing infill take far more time.

smartavionics, 2017-10-22:
Please see this [video](https://youtu.be/dUkSl2l1qs8)

It took 34 mins and didn't come out too bad, these show the part exactly as printed, I haven't touched it with a knife. 

[image]

[image]

This  shows there are some little bits of plastic on the top that could be trimmed but just a few seconds with a knife would remove them...

[image]

Thisismydigitalself, 2017-10-22:
Thank you , but haven't you just rested my case? :) 
these far too many clicks/retracts/points/jumps are soooo redundant.
I though this was simple to fix. i guess i was  wrong without really knowing why the algorithm can't overcome this somewhat simple challenge.

smartavionics, 2017-10-22:
> Thank you , but haven't you just rested my case? :)
these far too many clicks/retracts/points/jumps are soooo redundant.
I though this was simple to fix. i guess i was wrong without really knowing why the algorithm can't overcome this somewhat simple challenge.

Sorry, I don't agree. I'm not saying that what it does now is 100% optimal but with a part such as you have above, I cannot see how all travel moves can be avoided which it seems to me is what you are asking for. I don't believe it is possible to fill such a layer with lines without doing some travel moves.

Thisismydigitalself, 2017-10-22:
Well then.. let's agree to disagree and stay friends :)

Ghostkeeper, 2017-10-31:
We've tried spiral patterns but it's difficult to get right. In the end it's not quite going to be a spiral but simply concentric lines with a jagged corner somewhere where it moves to the next contour. Otherwise you'd have half-line-width gaps everywhere and the skin/infill just doesn't attach to the adjacent skin/infill or the walls.

And that concentric-with-jagged-corner pattern is pretty much the same as concentric anyway, like Smartavionics said, because in concentric it's a non-retracted travel move of 0.3mm that connects the adjacent contours. Whether you keep extruding for those 0.3mm or not, it's going to ooze out anyway so you're left with practically the same pattern. You'll still have these free-floating bits because they are actually just places where the nozzle is vibrating a bit in-place while extruding.

Thisismydigitalself, 2018-01-19:
Just checking if even thou this issue was closed it was addressed for future releases.
A week ago i have printed a very big sphere (in two halves) and it had thousands of small floating point(less) points like in the attached images above causing less than perfect print which lasted at least 30% longer than if they were addressed differently by Cura engine.

smartavionics, 2018-01-19:
Please upload the image to github as mail attachments are discarded.

Thisismydigitalself, 2018-01-20:
I will try to attach a video of my latest relevant print but in my previous post i was talking about these images in my earlier posts from October last year. this time my part had much more small bits of plastic which i felt sorry on behalf of my printer for letting it work so hard for nothing. of course some stl's , like this one i just printed, are far from being perfect and optimized for FDM printers but i do feel this artifact can be avoided.  what Ghostkeeper replied on Oct 31st made perfect sense to me but i was hoping there might be some good news regarding soon to be released 3.2.

So this post of mine is just checking if there was any progress on this matter.

smartavionics, 2018-01-20:
Hello. Sorry, there is lots of good news in 3.2 but not related to this issue. As @Ghostkeeper says above, when the walls are printed as groups (with the printing order optimization enabled) you are effectively getting a spiral (OK, a kinky spiral). The travel move from one wall to the next is often very short (the width of the line) and so the end of one wall and the start of the next will tend to merge. Are you sure the retraction settings are optimal for your printer, maybe the blobs can be reduced further?

Thisismydigitalself, 2018-01-20:
Yes, Retractions are OK, it's just there are so many of these taking place during printing when infill is set to 0 and enough walls to fill up the space between inner and outer walls. i tend to go this way as overall print time is shorter than when using infill and the outcome looks relatively better as there are no chaotic lines going in different directions (walls and infill in perpendicular manner) but more like a nice spirals which I am only trying to think out loud and see how Cura can further optimize those.

jackha, 2018-08-16:
By spiral patterns we mean concentric infill? Yes we have that :-)

Ghostkeeper, 2018-10-16:
No, we mean walls being connected together instead of concentric rings.