Ultimaker/Cura #2505: The contextual menu displays on mouse-up instead of mouse-down

yourhead, 2017-09-27:
All other menus in the app display as the user clicks the mouse down.  The contextual menu in the main work area displays on mouse-up.

I did a bit of a hunt. I couldn't find a single other instance of a **menu** popup that displayed on mouse-up. I did find several things that pop open a HUD -- this generally is on mouse up -- but menus...  always mouse-down.

macOS 10.12.6
Cura 2.7

ChrisTerBeke, 2017-09-28:
Currently the opening of context windows is bound to a mouse release event on every platform. On Windows (where the far majority of users is on as well), all my programs use mouse-up for showing context menus (or any right-click action). I think it's more a matter of personal preference than it being incorrect.

If you really want it changed, I invite you to take a look at https://github.com/Ultimaker/Uranium/blob/master/UM/Controller.py#L267 and create a pull request.

yourhead, 2017-09-28:
your team seems uninterested in user experience on this platform. i won’t waste my time with this group any more. you can close my other bugs if you like. :-)

nallath, 2017-09-28:
We are interested, the point is that there is more than just a single platform to think of. If we didn't care, we'd just close the issue right away instead of taking time to reply to them and explain why we have certain behavior.

I'm not a big fan of having to support multiple settings, which are then based on your operating system. Right now there is also the issue that the default rotate behavior is also on the right mouse, so if we were to show the context menu on mouse down, it will get really weird.

yourhead, 2017-09-28:
opinion: feel free to delete/ignore/ban. 

each issue that i've brought up, either here on GitHub, via support, or via twitter -- has been uniformly dismissed due to the needs of your dominant platform.

it's understandable that Windows take precedence. but goals for minimum user experience for other platforms should be set. if you can't support these minimum UI idioms and behaviors then you should probably drop the platform. supporting a platform in a second-rate way only does damage to your product.

that's not really a problem i can help solve. so i'll let you get on with your work and i'll get back to mine.  😃

closed 2017-09-28 by yourhead