Ultimaker/Cura #2506: Switch to AppImage Type 2

TheAssassin, 2017-09-28:
Currently, the Cura AppImages (just checked the 2.7.0 release) are type 1. This type is deprecated for good reasons. Type 2 has been around for a while, and aside from being more advanced and efficient, there's better tooling available which makes building them easier.

A few advantages:

 - More efficient due to use of squashfs instead of ISO 9660
 - Better ecosystem
   - especially important: [AppImageUpdate](https://github.com/AppImage/AppImageUpdate) -- allows for easy updating of AppImages (utilizing highly efficient binary delta updates) that ship with so-called [update information](https://github.com/AppImage/AppImageUpdate#example-how-to-make-your-app-self-updateable), I'd really recommend implementing this feature
   - [appimagetool](https://github.com/AppImage/AppImageKit/tree/appimagetool/master#appimagetool-usage), replacement for AppImageAssistant, faster and easier to use
   - [linuxdeployqt](https://github.com/probonopd/linuxdeployqt), a really easy to use advanced application for preparing AppDirs for any kind of application, not limited to Qt, but can also bundle Qt plugins with little to no effort
   - provides replacements for the hackish solutions in [this repository](https://github.com/AppImage/AppImages) which really shouldn't be used any more
 - Can be signed to provide authentication and integrity, although this hasn't been developed very far (help welcome, as usual)
 - More standalone (e.g., the AppImages can be extracted and manually mounted without additional tools)
 - ... and a few other features (@probonopd could provide some additional details)

I'd really recommend switching to type 2. If you need help, make sure to visit the AppImage IRC channel ([#appimage](https://webchat.freenode.net/?channels=appimage) on Freenode), we'll do our best to help.

You can check the version with this command:

```
> dd if=Cura-2.7.0.AppImage  bs=1 count=4 skip=8 | hexdump
0000000 4941 0001
                ^ version 1

# for comparison
> dd if=AppImageUpdate-x86_64.AppImage bs=1 count=4 skip=8 | hexdump
0000000 4941 0002
                ^ version 2
```

awhiemstra, 2017-09-28:
Back when we did the initial AppImage work, the v2 format was only a discussion item. Good to hear that things have been progressing there. Note that the https://github.com/ultimaker/cura-build repository is responsible for creating the AppImages. It should be fairly straightforward to upgrade this process to create v2 images.

TheAssassin, 2017-09-28:
Ah, okay. Yeah, I was looking for that configuration already. I'm going to create an issue there.

probonopd, 2017-11-12:
Clarification, we since moved the [AppImageKit](https://github.com/AppImage/AppImageKit) project to the [AppImage](https://github.com/AppImage) organization on GitHub, so the correct URL now is:

https://github.com/AppImage/AppImageKit/releases/tag/continuous

TheAssassin, 2018-01-31:
Posted a follow-up in the cura-build issue. The build system is now based on CMake. See https://github.com/Ultimaker/cura-build/issues/136 for details.

probonopd, 2018-01-31:
Should we close it here, then? @awhiemstra

Ghostkeeper, 2018-02-26:
Yeah let's close this in favour of https://github.com/Ultimaker/cura-build/issues/136

closed 2018-02-26 by Ghostkeeper