Ultimaker/Cura #2510: Unable to compile and run Cura from Source.

abinashk-inf, 2017-09-28:
I want to set up a developer instance of Cura. Using cura-build gives out following error:
/cura-build/build$ cmake ..
CMake Error at CMakeLists.txt:23 (get_filename_component):
  get_filename_component unknown component CACHE

CMake Error at projects/Uranium.cmake:5 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" (requested
  version 5.6.0) with any of the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:61 (include)

-- Configuring incomplete, errors occurred!
See also "/cura-build/build/CMakeFiles/CMakeOutput.log".` 
I have Qt5 already installed using Qt online installer. 
System Spec: Ubuntu 16.04 32bit

smartavionics, 2017-09-28:
Personally, I don't use cura-build. I first build/installed cura-build-environment and then just built/installed the individual components (Cura, CuraEngine, Uranium, etc.) separately.

abinashk-inf, 2017-09-29:
Hi I have installed all individual components of Cura(Uranium, CuraEngine). How do I launch the Cura ui? 
Can you let me know how did you set up your instance.
My aim is to launch Cura ui, make a few changes in code and see that changes reflect back

smartavionics, 2017-09-29:
Hi. I installed everything under /opt/cura (on a Linux system) and have a shell script (cura.sh) that looks like this:
````
#!/bin/sh
PATH=/opt/cura/bin:$PATH PYTHONPATH=/opt/cura/lib/python3/dist-packages /opt/cura/bin/cura $*
````
It works for me!

abinashk-inf, 2017-10-03:
Hi @smartavionics 
Thanks for the help. The cura-build was not really needed. For anyone who had problems with PyQt5 module because of missing PyQt5.QtQml they should install QtQuick module from archive. QtQuick has QtQuick and QtQml modules.
Closing the issue.

closed 2017-10-03 by abinashk-inf