Ultimaker/Cura #2511: Cura (master) does not slice

mahowi, 2017-09-28:
I'm using the builds (master) from @thopiekar as there a no more official builds for 32bit Linux.

As I have already reported in [issue #27](https://github.com/thopiekar/Cura-packaging/issues/27#issue-260736107) at Cura-packaging:
After loading a model to print it says "Slicing..." but that's it. It does not slice, no more lines in the logfile. Logfile is attached.
[cura.log](https://github.com/thopiekar/Cura-packaging/files/1334625/cura.log)

Slicing works in 2.6.1.

smartavionics, 2017-09-28:
The problem you have is indicated by this line in your cura.log:

2017-09-26 21:02:34,981 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'top_skin_expand_distance'

What this means is that CuraEngine is using a configuration value that isn't being set by the UI. This is almost certainly because the UI code and the CuraEngine code were not synchronised when the PPA was built. This can happen occasionally when people add/remove settings. Tomorrow it may be OK again. One of the perils of taking a snapshot of multiple repositories.

smartavionics, 2017-09-28:
Of course, taking a snapshot of a single repository isn't safe either because someone may be in the middle of making multiple commits.

ChrisTerBeke, 2017-10-04:
Probably caused by differences in Cura and CuraEngine that were introduced recently. It might be needed to re-compile CuraEngine to make it work with the latest Cura changes.

Ghostkeeper, 2017-10-15:
In this case, it seems CuraEngine is further ahead than Cura. You get that error when CuraEngine requests a setting that doesn't exist in the front-end.

thopiekar, 2017-10-15:
Well, before builds were failing because of the creation of a temp file outside the build environment, what launchpad.net does not allow I guess. I'm now ignoring all errors and got my fingers crossed that all errors are found via Ultimaker CI.

I guess this error does not appear anymore, because all packages are now built.
The only issue that is left is a segmentation fault, I get now. But this has been reported to a different issue ticket already.

closed 2018-02-26 by ianpaschal