Ultimaker/Cura #2515: Touch/pen support in windows?

clomads, 2017-09-29:
It'd be nice to have basic touch support for navigating the 3D space. I use Cura mainly on a Surface Pro so pen support would be nice too. Ultimately I really would just like to see some pinch to zoom. I can orbit with the pen using the 'left click' button, but being able to zoom without having to go to the trackpad (which I end up having detached with the keyboard sometimes) would be an immense improvement to the UI.

Thanks

fieldOfView, 2017-09-29:
Afaik, you can zoom with the scroll-gesture (two fingers up/down). Other than that I agree better touch support would be nice

clomads, 2017-09-29:
Yes, but that doesn't work on the touch screen... only the trackpad. In fact, if I have my touch cover thing (keyboard and trackpad cover) removed from the Surface Pro, I have no way to zoom in at all from what I can find.

fieldOfView, 2017-09-29:
Sorry, you are right. Doesn’t work on the touchscreen.

clomads, 2017-10-01:
It's all good.

Ghostkeeper, 2018-01-01:
Our project manager just rejected this issue because it's older than 12 weeks.

clomads, 2018-01-02:
So no response from the team? Am I just asking questions to a brick wall?

Ghostkeeper, 2018-01-02:
Mostly silicone and plastic wall.

I agree with you that this would be a good enhancement to have, so I put it up for consideration on the 5th of October. Last night it got taken off there because it got too old. Our project manager wants to keep the list small so that his boss doesn't see that there are still 1200 things to do, so he deletes stuff that is old. His argument is that there are likely not many people needing the feature if it didn't get updated in 12 weeks.

In effect, it means that we're not going to implement it anytime soon.

fieldOfView, 2018-01-02:
Someone from the community might pick it up...

nallath, 2018-01-03:
It's not from a lack of not wanting to pick this up, but we only have way more feature requests than we have people that can actually build them.

fufs, 2020-05-30:
It's 2020 and still no touchscreen support. Hell, the touchpad is also bad. Someone should do something about it!

nallath, 2020-06-02:
Correct. But my previous statement from ~2.5 years ago is still valid. It's not that we don't _want_ to fix it, it's just that there is too much to do and not enough people to fix it.

kaixxx, 2022-01-06:
Begging for touch support in 2022, please! You have a lot of other things to do, I know. But this is such a basic question of usability. We don't need any fancy features, just a very simple touch support. Right now I can't even open a menu (it's closing immediately) or selecting an object via the touchscreen.

auditor3d, 2022-04-25:
Touchscreen support is the only reason I keep SuperSlicer handy...Slic3r handles this differently than Cura and while I don't think it was intentional it does work on tablets. lol Mainly because the click and drag functions are the left mouse button instead of right.

fieldOfView, 2022-04-25:
> the left click and drag doesn't seem to do anything

Depending on the tool you have active and if you are starting the drag operation over the currently selected model, dragging will move/scale/rotate the current model. That's not exactly "nothing".

Clicking/dragging on an empty part of the viewport indeed does nothing. However, limiting the camera rotation to left-clicks started on empty parts of the viewport would mean you can no longer rotate the camera when you are zoomed in and/or handling a large model.

If it were a simple fix, I would have already created a PR for it (because I do agree with you that better touchscreen support is needed).

joseedil, 2022-08-17:
Issues #12353 and #11949 were routed here, but it doesn't seem like the same thing. Wacom tablets take place of the mouse and touchpads, not touchscreens.

Anyhow, I'm another one with Cura getting unresponsive after 4 or 5 seconds when using a Wacom tablet, but working properly with the Apple Touchpad. What's weird is that the software does work for a few seconds, but something goes wrong then.

MariMakes, 2022-08-23:
I agree with you @joseedil! 
Touch screen issues and Wacom tablets seem like different issues to me.
I'll move your tickets over to https://github.com/Ultimaker/Cura/issues/12554, that seems like a better fit. 😄

MrDB42, 2023-01-16:
I agree, touch screen is becoming more and more of a standard on computers (laptops, desktop screens, etc.). Just the ability to touch on value to be able to edit it, would be GREATLY appreciated.

3DStreetFurniture, 2023-01-31:
I have a 15.6" touch screen to a NUC that has Cura 5.2.1 and zoom, rotate and the general click and move the camera around does not work with touch screen. Any idea on when this will be supported?

melyux, 2025-04-04:
Pinch to zoom is just table stakes at this point. Makes Cura feel like a janky app on Mac.