Ultimaker/Cura #2516: Cura 2.7 loses settings between relaunches

MarioStoilov, 2017-09-29:
From time to time, whenever I launch cura, I find that either my printer profile is missing, or my filament profile, or my slicing profile. This is somewhat inconsistent and I cant determine the steps needed to reproduce, but its happening quite frequently.

nallath, 2017-09-29:
Could you share your logs? If possbile; Could you also share a project file that you saved after it went wrong?

MarioStoilov, 2017-09-29:
Well I do not do projects, I just load up a model, slice it, send to Octoprint and close cura.

For the logs, I'm going t need a location for those.

nallath, 2017-09-29:
You can find it under help -> show config locations.

If you can save a profile for me, i can help you better. It also contains information that I can use to diagnose the problem.

MarioStoilov, 2017-10-01:
[Cura.zip](https://github.com/Ultimaker/Cura/files/1347152/Cura.zip)

These are the profiles I use . The printer one just got lost again. It was after I booted my machine.

nallath, 2017-10-02:
It seems the machine that you are using has been corrupted somehow. You could try deleting it and re-adding it again.

closed 2018-03-02 by ChrisTerBeke